tinygo-org/tinygo #1973: Design for SPI using DMA with atsamd51

sago35, 2021-06-28:
I am working on a design for SPI with DMA on atsamd51.

> I think we'll always want to use DMA. That is an easier to use API and likely faster overall.
> https://github.com/tinygo-org/tinygo/pull/1176#issuecomment-648437757

Tx() to always use dma if possible, but at least atsamd51 slows things down in some cases.
If the data is small or the communication speed is slow, using DMA will slow down spi.Tx().

To reduce the slowdown, I am considering implementing the following.
If anyone has any opinions, please let me know.

```go
// The following will be executed with blocking as before
// The internal implementation will be the same as before, without DMA.
spi.Tx(w, r)

// The following will be executed using DMA.
// It will always run asynchronously and can wait for completion by calling spi.
spi.TxDMA(w, r)
spi.Wait()

```

sago35, 2021-06-28:
D6 (Channel 6) below is the processing time for spi.Tx() / spi.TxDMA().
When using DMA, about 2.5us of time is needed to prepare for the transfer.
However, after 2.5us has elapsed, the DMA will execute, allowing you to perform other operations.
I checked several lengths, and it seems that using DMA is not a problem when the length is 32 bytes or more.

freq : 24MHz

1byte + without DMA = 738ns
[image]

1byte + with DMA = 4446ns
[image]

32byte + without DMA = 13.494us
[image]

32byte + with DMA = 14.798us
[image]

64byte + without DMA = 26.638us
[image]

64byte + with DMA = 25.452us
[image]

soypat, 2023-11-06:
I'm linking a implementation of DMA on the RP2040 for your viewing pleasure. It is inspired by Rust's embassy implementation.

https://github.com/tinygo-org/pio/blob/main/rp2-pio/piolib/dma.go

sago35, 2023-11-08:
I plan to switch to the method used in the following PR.

#3985

closed 2023-11-08 by sago35