tinygo-org/tinygo #1981: panic: runtime error: nil pointer dereference

fgsch, 2021-07-01:
Full error:

```
panic: runtime error: nil pointer dereference
error: failed to run compiled binary /some/path/main: signal: abort trap
```

TinyGo version:

`tinygo version 0.19.0 darwin/amd64 (using go version go1.16.5 and LLVM version 11.0.0)`

Command line:

`tinygo run main.go`

Repro:

```go
package main

import "flag"

var _ = flag.Bool("foo", false, "")

func main() {
	flag.Parse()
}
```

Also fails with TinyGo 0.17.0 and 0.18.0.

Originally found in https://github.com/tinygo-org/tinygo/pull/1814. Reporting it here so it can be tracked independently.

fgsch, 2021-07-01:
Backtrace:

```
#1  0x00007ffff7df3859 in __GI_abort () at abort.c:79
#2  0x00000000004014f6 in runtime.runtimePanic (msg=...) at /home/fgsch/tinygo/src/runtime/panic.go:20
#3  0x00000000004017f4 in runtime.nilPanic () at /home/fgsch/tinygo/src/runtime/panic.go:32
#4  0x0000000000401c10 in (*flag.FlagSet).Var (f=0x0, value=..., name=..., usage=...) at /opt/go-1.16.4/src/flag/flag.go:862
#5  0x0000000000401a73 in runtime.initAll () at /home/fgsch/tinygo/src/runtime/runtime.go:33
#6  0x0000000000401a42 in runtime.run$1 () at /home/fgsch/tinygo/src/runtime/scheduler_any.go:19
#7  0x0000000000401a2c in <goroutine wrapper> () at /home/fgsch/tinygo/src/runtime/scheduler_any.go:18
#8  0x0000000000401c31 in tinygo_startTask () at /home/fgsch/tinygo/src/internal/task/task_stack_amd64.S:29
```

fgsch, 2021-07-11:
Similar crash if running under wasi:

```
panic: runtime error: nil pointer dereference
error: failed to run `/some/path/tinygo172976275/main`
│   1: WASI execution failed
│   2: failed to run WASI `_start` function
│   3: RuntimeError: unreachable
           at runtime.runtimePanic (main[13]:0xe48)
           at runtime.nilPanic (main[21]:0xfaf)
           at (*flag.FlagSet).Var (main[37]:0x1459)
           at runtime.initAll (main[36]:0x140c)
           at runtime.run$1 (main[35]:0x1386)
           at runtime.run (main[34]:0x137c)
           at _start (main[33]:0x136c)
╰─> 4: unreachable
```

aykevl, 2021-07-14:
This bug is fixed with this PR: https://github.com/tinygo-org/tinygo/pull/1999

dyaskur, 2021-08-29:
@aykevl Does your PR also fix my problem?

on tinygo 1.18, my code work fine, but on 1.19 why is still return:

```
panic: runtime error: nil pointer dereference
0002035a:0x13ad Uncaught RuntimeError: unreachable
    at runtime.runtimePanic (<anonymous>:wasm-function[33]:0x13ad)
    at runtime.nilPanic (<anonymous>:wasm-function[15]:0xba3)
    at (io.Reader).Read (<anonymous>:wasm-function[26]:0x10c6)
    at io.ReadAtLeast (<anonymous>:wasm-function[25]:0x10bc)
    at io.ReadFull (<anonymous>:wasm-function[27]:0x10ea)
    at github.com/mervick/aes-everywhere/go/aes256.Encrypt (<anonymous>:wasm-function[99]:0x6f96)
    at syscall/js.handleEvent.resume (<anonymous>:wasm-function[91]:0x5cd0)
    at runtime.scheduler (<anonymous>:wasm-function[64]:0x35ee)
    at resume (<anonymous>:wasm-function[70]:0x3ae1)
    at global.Go._resume (vendor.js:7901)
```

 @fgsch  (issuer) said on 1.18 failed too, but my code worked fine on 1.18. below is my code:
```

func getCode(this js.Value, i []js.Value) interface{} {

	value1 := i[0].String()

	encrypted := aes256.Encrypt(value1, "Phrase")

	return strings.Replace(encrypted, "U2FsdGVkX1", "", 1)
}
```

Thanks

aykevl, 2021-09-05:
@dyaskur I don't think you're hitting this bug, but instead https://github.com/tinygo-org/tinygo/issues/2020 which is fixed in the dev branch and will be part of the next release.

deadprogram, 2021-09-07:
Since #2021 has been merged perhaps @dyaskur you can try with the latest `dev` branch.

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram