tinygo-org/tinygo #1985: stm32: some timers (PWM) cannot output to pin

kenbell, 2021-07-05:
The stm32l032 for example does not enable pin output.  The root cause is that I assumed no BDTR register meant no need to enable pin output.

However, the BDTR register was just missing from the Go code generated by the `gen-device` tool.

### Root Cause

The BDTR field is missing because:

1. not all TIM peripherals have all registers present (however they do have 'compatible' layout) - so if a particular register is present in a peripheral it WILL be at the same address on all peripherals
2. `gen-device` assumes all peripherals in a 'group' are identical, so it takes the first and uses it's register layout as definitive
3. in some cases, gen-device first see's a TIM peripheral with the BDTR register not present

kenbell, 2021-07-05:
I'm working on a fix, raising this issue since there's going to be a few PR linked to fixing this.

aykevl, 2021-07-06:
> The BDTR field is missing because:
>
>     1. not all TIM peripherals have all registers present (however they do have 'compatible' layout) - so if a particular register is present in a peripheral it [...]

Right. Yes, so far it seemed like a groupName would just bundle a few similar or identical peripherals together, I didn't expect them to vary in the actual layout. But maybe I shouldn't have had such expectations with vendor-supplied data.

This is the official specification, it seems: https://www.keil.com/pack/doc/CMSIS/SVD/html/svd_Format_pg.html

That said, I can't find a BDTR field in the stm32l0x2 SVD file? Or is it just named differently?

kenbell, 2021-07-06:
> Peripherals that provide similar functionality (Simple Timer, Complex Timer) can be grouped with the element <groupName>. Peripheral groups help structuring the list of peripherals in the debugger. All peripherals associated with the same group name are collectively listed under this group in the order they were specified in the SVD file.

From there doc it seems like the grouping is meant for visual layout in a debugger.  One of the SVDs has `ADC1`, `ADC2`, `ADC_Common` all in the `ADC` group :)

deadprogram, 2022-04-28:
@kenbell @aykevl was this ever addressed?