tinygo-org/tinygo #1988: runtime/volatile: `register.ReplaceBits` helpers misplaced space at `//go:inline` pragma

soypat, 2021-07-09:
The following `//go:inline` pragmas seem to have unwanted indentation that could potentially prevent their inlining and more certainly cause confusing documentation generation.

- https://github.com/tinygo-org/tinygo/blob/release/src/runtime/volatile/register.go#L127
- https://github.com/tinygo-org/tinygo/blob/release/src/runtime/volatile/register.go#L190
- https://github.com/tinygo-org/tinygo/blob/release/src/runtime/volatile/register.go#L251
- https://github.com/tinygo-org/tinygo/blob/release/src/runtime/volatile/bitband_nxpmk66f18.go#L54
- https://github.com/tinygo-org/tinygo/blob/release/src/runtime/volatile/bitband_nxpmk66f18.go#L64
- https://github.com/tinygo-org/tinygo/blob/release/src/runtime/volatile/bitband_nxpmk66f18.go#L74

These too have a space, though it may be on purpose:
- https://github.com/tinygo-org/tinygo/blob/release/compiler/symbol.go#L24-L31

deadprogram, 2023-05-23:
A pull request to correct these maybe? I notice they are still this way.

aykevl, 2023-05-26:
Yes, these should indeed be fixed!

> These too have a space, though it may be on purpose:
>
> * https://github.com/tinygo-org/tinygo/blob/release/compiler/symbol.go#L24-L31

Yes, those are normal comments and are intended that way.

PR #3755 (2023-05-29): Fix volatile inlines
Resolves #1988
closed 2023-06-14 by deadprogram