tinygo-org/tinygo #1990: Another memory corruption problem?

fgsch, 2021-07-09:
The code below prints:

`len(b)=55265: "foo" -> ""`

Tinygo version:

`tinygo version 0.19.0 darwin/amd64 (using go version go1.16.5 and LLVM version 11.0.0)`

Code:

```
package main

import (
	"fmt"
	"os"
)

func main() {
	pre := os.Getenv("PATH")
	b := make([]byte, 55265)
	post := os.Getenv("PATH")
	fmt.Printf("len(b)=%d: %q -> %q\n", len(b), pre, post)
}
```

Compilation and testing:

- `tinygo build -target=wasi -o bug.wasm bug.go`
- `wasmtime run  bug.wasm --env PATH=foo`

This might be a duplicate of #1966.

dkegel-fastly, 2021-07-09:
Thanks for finding a minimal reproducer!

aykevl, 2021-07-10:
Indeed, this minimal reproducer made it much easier to spot the underlying issue!

I think I know the bug, this should be a fix: https://github.com/tinygo-org/tinygo/pull/1992
I'm not sure whether this is related to #1966 but if you can test it, that would be great.

fgsch, 2021-07-10:
Thanks. #1992 fixes the issue in this PR but #1966 still crashes.

fgsch, 2021-07-14:
Credits should go to @peterbourgon for repro.

fgsch, 2021-07-14:
This can be marked as next-release now 🎉

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram