tinygo-org/tinygo #1996: mqtt cannot find package packets

mybigman, 2021-07-12:
```
package main

import (
	"machine"
	"time"

	"tinygo.org/x/drivers/espat"
	"tinygo.org/x/drivers/net/mqtt"
)
```

```
/go/src/tinygo.org/x/drivers/net/mqtt/mqtt.go:10:2: cannot find package "github.com/eclipse/paho.mqtt.golang/packets" in any of:
	/app/.cache/tinygo/goroot-go1.15.13-9644191d5736a3bd1d629cebaee90a87d6f036d703b2e69f4df9bcc262d266df/src/github.com/eclipse/paho.mqtt.golang/packets (from $GOROOT)
	/go/src/github.com/eclipse/paho.mqtt.golang/packets (from $GOPATH)
package command-line-arguments
	imports tinygo.org/x/drivers/net/mqtt
```

deadprogram, 2021-07-16:
You probably need to run `go get github.com/eclipse/paho.mqtt.golang` to add to the modules for your project executable.

aykevl, 2021-07-16:
It looks like you're not using Go modules, is that correct?
The easiest fix is probably `go mod init` and then adding the module as @deadprogram indicated.

deadprogram, 2023-05-23:
Closing due to age. Please reopen if needed, thanks!

closed 2023-05-23 by deadprogram