tinygo-org/tinygo #1997: Bug: package runtime/trace is not in GOROOT

omar391, 2021-07-12:
I am having this issue while trying to compile my program to -target wasm:

`tinygo build -o ./nodejs/lib-tiny.wasm -target wasm ./main.go`

It throws this error:
`C:\Users\**\go\pkg\mod\go.opencensus.io@v0.23.0\trace\trace_go11.go:21:2: package runtime/trace is not in GOROOT (C:\Users\**\AppData\Local\tinygo\goroot-go1.16.5-b6761f345c63a2cb198265aef5bcc7da1c2af2c14394faa25cfb608af5b42a9c\src\runtime\trace)`

Is there a way to print more build error stack trace info and any solution for the `package runtime/trace` issue?

Similar issues in another repo: https://github.com/benthosdev/benthos-lab/issues/22
Best.

wolverian, 2021-12-09:
This also breaks OpenTelemetry unfortunately:

```
/Users/ii/go/pkg/mod/go.opentelemetry.io/otel/sdk@v1.2.0/trace/span.go:22:2: package runtime/trace is not in GOROOT (/Users/ii/Library/Caches/tinygo/goroot-go1.17.3-0235e5915a10fdaa87b6f2130d112a9a9ae3c9c24c9419c23e81c7c218fa6161-syscall/src/runtime/trace)
```

dgryski, 2021-12-09:
I don't think it would be difficult to build a tinygo-specific `runtime/trace` package that exports the same API, but obviously the functionality would be minimal as the tinygo runtime doesn't actually support the required tracing features.  That would allow these packages to build, but not run with the identical feature set.

joeshaw, 2022-09-08:
This is fixed by #3117 but I don't have perms to close this ticket.

PR #3117 (2022-09-02): runtime/pprof, runtime/trace: stub some additional functions
These are necessary to import some parts of the net/http package.

Fixes #1997
closed 2022-09-12 by omar391