tinygo-org/tinygo #2000: goroutine stack size

bradleypeabody, 2021-07-15:
I couldn't find anything in the docs about the stack memory requirements of each goroutine created in TinyGo on a microcontroller (SAMD21 in my case).

Context: I'm specifically wondering if I make, let's say a dozen, goroutines each with different loops such as:
```
go func() {
    for {
        time.Sleep(/* some duration appropriate to this action */)
        // some action
    } 
}
```
How much memory is wasted just in maintaining the stack for each?  I realize these are translated into LLVM coroutines, but I imagine the stack has to go somewhere and would need to be allocated when the goroutine starts.  If each stack is 2k, then 12 of these uses 24k of memory which is probably a bad way to go and I'd be better of scheduling these actions some other way.  (I realize too many sleeps like this with various actions in between would make the timing of each sleep less precise as they are all sharing one thread of execution, but at the moment I'm more concerned about the memory usage).

So my specific question is, **how much stack memory is used for each goroutine?**

aykevl, 2021-07-15:
In short: it depends. It varies by chip. Recently the default stack size was increased from 1kB to 2kB for most chips (see https://github.com/tinygo-org/tinygo/pull/1920 and https://github.com/tinygo-org/tinygo/pull/1960). However, the compiler might be able to determine the maximum stack size at compile time and will then use this value instead. You can see whether it works with the `-print-stacks` command line flag. For example:

```go
package main

import "time"

func main() {
	go func() {
		for {
			time.Sleep(time.Second)
			// some action
		}
	}()
	println("test")
	time.Sleep(time.Minute)
}
```

```
$ tinygo build -o test.uf2 -target=itsybitsy-m0 -print-stacks issue2000.go
function                         stack usage (in bytes)
Reset_Handler                    recursive, runtime.runtimePanic may call itself
main.main$1                      recursive, runtime.runtimePanic may call itself
runtime.run$1                    recursive, runtime.runtimePanic may call itself
```

In this case, it isn't able to determine the stack size automatically. I believe this is because of the USB-CDC code which is rather complicated and the compiler can't figure out the maximum stack size for it. For example, it works on this board that doesn't use the USB-CDC output:

```
$ tinygo build -o test.uf2 -target=pca10040 -print-stacks tmp/issue2000.go
function                         stack usage (in bytes)
Reset_Handler                    324
main.main$1                      100
runtime.run$1                    284
```

As you can see, the goroutine that only calls `time.Sleep` in a loop takes up just 100 bytes of stack space (plus some extra metadata).

> I realize these are translated into LLVM coroutines

They are not. This is only the case if you use `-scheduler=coroutines` or if you are using WebAssembly. For all other platforms, it will default to the tasks scheduler which is very similar to the regular Go runtime.

bradleypeabody, 2021-07-15:
Perfect, thank you.  I'll definitely use `-print-stacks`.  And I see now that `default-stack-size` is specified in the target file, e.g. `targets/cortex-m.json`

closed 2021-07-15 by bradleypeabody