tinygo-org/tinygo #2006: Go 1.17 compabability

deadprogram, 2021-07-19:
Has anyone looked into the upcoming Go 1.17 release, and if TinyGo will require any changes to accommodate it?

aykevl, 2021-07-25:
I looked at the release notes and there are in fact a few changes to the language which likely will need to be supported in TinyGo too. The `unsafe` changes probably don't have an effect (except that TinyGo may need to be built with Go 1.17 to support Go 1.17), but the new slice to array cast will likely require a change in TinyGo.

https://tip.golang.org/doc/go1.17#language

aykevl, 2021-07-25:
I intend to wait a bit until the dust is settled on the x/tools/go/ssa package: https://github.com/golang/go/issues/46987

aykevl, 2021-07-25:
Also, the math package got some problematic changes that I'm not sure how to solve.
In other words, supporting Go 1.17 will unfortunately be a bit more work.

oflebbe, 2021-08-05:
@aykevl Just for my curiosity : Regarding math I see only the MaxUInt etc constants mentioned, which IMO doesn't sound too alarming to me. Can you go into more detail what might be problematic to solve ?

aykevl, 2021-08-05:
The problem is this commit: https://github.com/golang/go/commit/1d20a362d0ca4898d77865e314ef6f73582daef0. Previously, it was simple to forward the math functions to the naive/non-optimized versions. But with that commit, it becomes a bit more complicated. (This is why I started work on supporting the optimized assembly versions).

oflebbe, 2021-08-05:
Any chance to have [FPU](https://github.com/tinygo-org/tinygo/commit/f79e66ac2e66a5338ec3ce3c4c992567b07c4959) reenabled again , or did I miss it ? Otherwise assembly will be really hard.

aykevl, 2021-08-05:
> Any chance to have [FPU](https://github.com/tinygo-org/tinygo/commit/f79e66ac2e66a5338ec3ce3c4c992567b07c4959) reenabled again

This is only for the Cortex-M. The FPU is always enabled on desktop systems (386, amd64, arm64). The assembly support I'm working on will only be available for targets supported upstream (running on Linux and MacOS), not for any other architectures. It's not possible (or very hard) to automatically convert aarch64 (arm64) machine code to T32 (32-bit ARM, Cortex-M) machine code.

oflebbe, 2021-08-05:
Ah interesting. 25years ago I did an 387 FPU binding for the linux libm :)
But why do you an assembly implementation on your own ? Shouldn't the llvm.sin.* etc intrinsic work quite good like you already implemented for math.Sqrt ?

aykevl, 2021-08-05:
>  But why do you an assembly implementation on your own ? Shouldn't the llvm.sin.* etc intrinsic work quite good like you already implemented for math.Sqrt ?

I tried, but sadly this doesn't work. When I do this, the tests in the math packages don't pass.
Note that I'm not writing the assembly implementations myself. I'm writing code that will make use of the [sqrt_amd64.s](https://github.com/golang/go/blob/release-branch.go1.16/src/math/sqrt_amd64.s) code that already exists, for example:

```
// func Sqrt(x float64) float64
TEXT ·Sqrt(SB), NOSPLIT, $0
	XORPS  X0, X0 // break dependency
	SQRTSD x+0(FP), X0
	MOVSD  X0, ret+8(FP)
	RET
```

oflebbe, 2021-08-06:
Hm, for me the math package works in #2034 with llvm builtins. However the min() and max() functions had been incorrectly implemented for me. This approach has the advantage that llvm will even compile constants . What do you think?

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram