tinygo-org/tinygo #2012: got LinkErrors when i run wasm on browsers

Taoja, 2021-07-23:
the error said:
`LinkError: WebAssembly.instantiate(): Import #5 module="env" function="math/big.shrVU" error: function import requires a callable`

golang version: 1.14.6
tinyGo version: 0.19.0
wasm_exec.js is come from tinyGoRoot/targets/
build shell command: 
`tinygo build -o pkg/lib.wasm -target wasm main.go`

fgsch, 2021-07-23:
This should be fixed by #2002. 
In the meantime try building with `tinygo build -target wasm -tags math_big_pure_go -o ... main.go`.

Taoja, 2021-07-26:
benchmark error : 
`Benchmark not declared by package testing`
when i run
`tinygo build -o fibo ./fibo.go`

> This should be fixed by #2002.
> In the meantime try building with `tinygo build -target wasm -tags math_big_pure_go -obj pkg/lib.wasm main.go`.

deadprogram, 2024-12-02:
Closing since it appears to have been fixed. Please reopen if needed. Thank you!

closed 2024-12-02 by deadprogram