tinygo-org/tinygo #2019: wasm got runtime error

Taoja, 2021-07-30:
> This should be fixed by #2002.
> In the meantime try building with `tinygo build -target wasm -tags math_big_pure_go -obj pkg/lib.wasm main.go`.

the -obj flag is error? 
`flag provided but not defined: -obj`

i try change the -obj to -o then it is worker.
but got a new error on runtime:
`
syscall/js.finalizeRef not implemented
syscall/js.finalizeRef @ lib.js:305
syscall/js.jsString @ lib.wasm:0xf24f
(syscall/js.Value).String.resume @ lib.wasm:0xeefc
runtime.scheduler @ lib.wasm:0xcfe5
resume @ lib.wasm:0xde10
_resume @ lib.js:490
(anonymous) @ lib.js:501
syscall/js.valueNew @ lib.js:380
(syscall/js.Value).New.resume @ lib.wasm:0x103f3
runtime.scheduler @ lib.wasm:0xcfe5
resume @ lib.wasm:0xde10
_resume @ lib.js:490
(anonymous) @ lib.js:501
(anonymous) @ VM626:1

lib.js:267 panic: runtime error: nil pointer dereference

VM626:2 RuntimeError: unreachable
    at runtime.runtimePanic (lib.wasm:0x957b)
    at runtime.nilPanic (lib.wasm:0x205e)
    at (io.Reader).Read (lib.wasm:0x62b7)
    at io.ReadAtLeast (lib.wasm:0x6265)
    at io.ReadFull (lib.wasm:0x6390)
    at github.com/tjfoc/gmsm/sm2.randFieldElement (lib.wasm:0x3bb46)
    at github.com/tjfoc/gmsm/sm2.Encrypt (lib.wasm:0x3a7bf)
    at main.encryptKey.resume (lib.wasm:0x4165c)
    at runtime.scheduler (lib.wasm:0xcfe5)
    at resume (lib.wasm:0xde10)
`

_Originally posted by @Taoja in https://github.com/tinygo-org/tinygo/issues/2012#issuecomment-886335870_

fgsch, 2021-07-30:
Yes, sorry, `-o` is correct (and not `-obj`).

Taoja, 2021-08-25:
> Yes, sorry, `-o` is correct (and not `-obj`).

`package main

import (
	cRand "crypto/rand"
	"encoding/base64"
	"errors"
	"github.com/tjfoc/gmsm/sm2"
	"github.com/tjfoc/gmsm/sm4"
	"github.com/tjfoc/gmsm/x509"
	"math/rand"
	"syscall/js"
	"time"
)

var kvMap = make(map[string] []byte) // 内存键值对

// 加密结果
type encryptResult struct {
	enc string // 加密密文base64格式
	key string // 加密秘钥base64格式
	err error // 错误信息
}

// 解密结果
type decryptResult struct {
	dec []byte // 解密后字节码
	err error // 错误信息
}

// 主进程
func main() {
	c := make(chan bool)
	vm := js.Global().Get("vm")
	jsEncrypt := js.FuncOf(doEncrypt) // 创建js加密函数
	jsDecrypt := js.FuncOf(doDecrypt) // 创建js解密函数
	vm.Set("encrypt", jsEncrypt) // 加密函数赋值给window.encrypt
	vm.Set("decrypt", jsDecrypt) // 解密函数赋值给window.decrypt
	<- c
}

// 暴露给js的加密方法
func doEncrypt(_ js.Value, data []js.Value) interface{} {
	Promise := js.Global().Get("Promise")
	callback := js.FuncOf(func(_ js.Value, cb []js.Value) interface{} {
		
		Error := js.Global().Get("Error")
		resolve := cb[0]
		reject := cb[1]
		
		// 判断参数是否正确
		if len(data) < 1 {
			reject.Invoke(Error.New("参数不足"))
			return nil
		}
		
		channel := make(chan encryptResult, 1)
		// 对内容进行sm4加密
		go doSm4Encrypt(data[0].String(), channel)
		result := <- channel
		if result.err != nil {
			reject.Invoke(Error.New(result.err.Error()))
			return nil
		}
		
 		resolve.Invoke([]interface{}{result.enc, result.key})
		return nil
	})
	return Promise.New(callback)
}

// 暴露给js的解密方法
func doDecrypt(_ js.Value, data []js.Value) interface{} {
	Promise := js.Global().Get("Promise")
	callback := js.FuncOf(func(_ js.Value, cb []js.Value) interface{} {
		
		Error := js.Global().Get("Error")
		resolve := cb[0]
		reject := cb[1]
		
		// 判断参数是否正确
		if len(data) < 2 {
			reject.Invoke(Error.New("参数不足"))
			return nil
		}
		
		channel := make(chan decryptResult, 1)
		// 对内容进行sm4加密
		go doSm4Decrypt(data[0].String(), data[1].String(), channel)
		result := <- channel
		if result.err != nil {
			reject.Invoke(Error.New(result.err.Error()))
			return nil
		}
		
		resolve.Invoke(string(result.dec))
		return nil
	})
	return Promise.New(callback)
}

// sm4加密报文
// @param data string 加密字符串
// @param channel chan 通道
// @return encryptResult
func doSm4Encrypt(data string, channel chan encryptResult) {
	result := encryptResult{}
	// 生成随机秘钥
	key := initRandomKey()
	
	pdt := []byte("HMBS") // 协商头
	pdb := []byte("HMBA") // 协商尾
	
	realKey := make([]byte, 16)
	
	copy(realKey[0:4], pdt)
	copy(realKey[4:12], key)
	copy(realKey[12:16], pdb)
	
	// 通过随机秘钥对内容进行加密
	enc, err := sm4.Sm4Ecb(realKey, []byte(data), true)
	if err != nil {
		result.err = err
		channel <- result
		return
	}
	encB64 := base64.StdEncoding.EncodeToString(enc)
	
	// 通过sm2加密随机秘钥
	keyB64, err := encryptKey(key)
	if err != nil {
		result.err = err
		channel <- result
		return
	}
	
	// 将密文秘钥与明文秘钥通过map进行存储
	setMap(keyB64, realKey)
	
	result.enc = encB64
	result.key = keyB64
	channel <- result
	return
}

// sm4解密报文
// @param data 需解密字符串base64编码
// @param key 秘钥密文base64编码
// @return decryptResult
func doSm4Decrypt(data string, key string, channel chan decryptResult) {
	// 创建结果
	result := decryptResult{}
	
	// 密文转byte数组
	dataByte, err := base64.StdEncoding.DecodeString(data)
	if err != nil {
		result.err = err
		channel <- result
		return
	}
	
	// 获取明文sm4秘钥
	keyByte, err := getMap(key)
	if err != nil {
		result.err = err
		channel <- result
		return
	}
	
	// 解密
	dec, err := sm4.Sm4Ecb(keyByte, dataByte, false) // 解密
	if err != nil {
		result.err = err
		channel <- result
		return
	}
	
	deleteMap(key)
	
	result.dec = dec
	channel <- result
	return
}

// 生成随机秘钥
func initRandomKey() []byte {
	m := []byte("1234567890qwertyuiopasdfghjklzxcvbnm")
	rand.Seed(time.Now().UnixNano())
	rand.Shuffle(36, func(i, j int) { m[i], m[j] = m[j], m[i] })
	return m[0:8]
}
// 使用sm2加密sm4秘钥
func encryptKey(key []byte) (string, error) {
	pub, err := x509.ReadPublicKeyFromHex("041bf86ecf5d8699d2b22d4eb88e7118ed8ba89ea2d0f45445ec87e72adfce064b61e98d88e616d412c9c939ffbc3b4797ab47775d72027d80e68109dfef97709f")
	if err != nil {
		return "", err
	}
	enc, err := sm2.Encrypt(pub, key, cRand.Reader, 0) //sm2加密
	if err != nil {
		return "", err
	}
	b64 := base64.StdEncoding.EncodeToString(enc)

	return b64, nil
}

// 创建sm4秘钥加密后base64与明文秘钥的对应关系
func setMap(keyB64 string, key []byte) {
	kvMap[keyB64] = key
}

// 通过base64秘钥查询明文秘钥
func getMap(keyB64 string) ([]byte, error) {
	key, err := kvMap[keyB64]
	if !err {
		return nil, errors.New("内存键值对检索失败")
	}
	return key, nil
}

func deleteMap(keyB64 string) {
	delete(kvMap, keyB64)
}
`

here is my code, do not know where i use syscall/js. finalizeRef

yaliv, 2021-08-25:
I got that error too in browser when I build with `-tags math_big_pure_go` option.
`syscall/js` mentioned there seems related to the `wasm_exec.js` file.
Look at these lines: https://github.com/tinygo-org/tinygo/blob/v0.19.0/targets/wasm_exec.js#L302-L306

dyaskur, 2021-08-29:
first you need edit syscall/js. finalizeRef like https://github.com/tinygo-org/tinygo/issues/1140 , but on error `panic: runtime error: nil pointer dereference`  I have that problem too, and I think it is similar to https://github.com/tinygo-org/tinygo/issues/1981 . hope PR https://github.com/tinygo-org/tinygo/pull/1999 will fix this problem too.

deadprogram, 2021-09-07:
Is this still an issue with the latest `dev` branch?

dyaskur, 2021-09-08:
Actually I want to try dev branch but dont know how to do that. Any references on how to build/install tinygo from repo? I dont see any documentation how to do it.

aykevl, 2021-09-08:
Here is the documentation: https://tinygo.org/docs/guides/build/

deadprogram, 2021-11-10:
This should have been fixed in the last release of TinyGo, so now closing. Please reopen if needed. Thanks!

closed 2021-11-10 by deadprogram