tinygo-org/tinygo #2020: Wasm crypto/rand Reader nil pointer dereference

jgbz, 2021-07-30:
In this new release, the package crypto/rand has been replaced with the tinygo version.
The golang original package, on Wasm, Reader uses the Web Crypto API, and that is not implemented.
I am getting an error when trying to encrypt/decrypt.

```
panic: runtime error: nil pointer dereference
main.wasm:0x4486 Uncaught (in promise) RuntimeError: unreachable
    at runtime.runtimePanic (main.wasm:0x4486)
    at runtime.nilPanic (main.wasm:0x1ab1)
    at (io.Reader).Read (main.wasm:0x419f)
    at io.ReadAtLeast (main.wasm:0x4195)
    at io.ReadFull (main.wasm:0x41c3)
    at main.main (main.wasm:0x9345)
    at runtime.run$1 (main.wasm:0x78f4)
    at _start (main.wasm:0x7433)
    at global.Go.run (wasm_exec.js:478)
    at wasm_init.js:194
```

This is the source code of the rand_js file that go uses to get the web crypto api

```// Copyright 2018 The Go Authors. All rights reserved.
// Use of this source code is governed by a BSD-style
// license that can be found in the LICENSE file.

// +build js,wasm

package rand

import "syscall/js"

func init() {
	Reader = &reader{}
}

var jsCrypto = js.Global().Get("crypto")
var uint8Array = js.Global().Get("Uint8Array")

// reader implements a pseudorandom generator
// using JavaScript crypto.getRandomValues method.
// See https://developer.mozilla.org/en-US/docs/Web/API/Crypto/getRandomValues.
type reader struct{}

func (r *reader) Read(b []byte) (int, error) {
	a := uint8Array.New(len(b))
	jsCrypto.Call("getRandomValues", a)
	js.CopyBytesToGo(b, a)
	return len(b), nil
}```

jgbz, 2021-07-30:
I put the rand_js.go file in the tinygo crypto/rand directory, and apparently it worked fine, im not getting the error anymore

aykevl, 2021-07-31:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/2021

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram