tinygo-org/tinygo #2023: wasm - "panic: unimplemented: alignment of type" on large fmt.Fprintf?

dkegel-fastly, 2021-08-02:
(This is from a later version of the app in https://github.com/tinygo-org/tinygo/issues/1826 )

With tinygo 0.19 from homebrew, building the app with

tinygo build -target=wasi -scheduler=coroutines -gc=leaking -wasm-abi=generic

and running its smoke test suite yields 36 panics of three flavors:
```
  17 panic: reflect: call of reflect.Value.Elem on invalid type
  11 panic: unimplemented: (reflect.Value).Pointer()
   1 panic: unimplemented: alignment of type
```
With today's dev ( 98e70c9b1967 ) or Ayke's gc-pass-deal-with-musttail, running the smoke test yields fewer panics, all of one sort:
```
   7 panic: unimplemented: alignment of type
```

which seems like progress :-) but may also just be luck, as I saw one more panic type while experimenting.

The log from the dev run says

```
panic: unimplemented: alignment of type
Aug 02 10:36:41.138 ERROR request{id=2}: WebAssembly trapped: wasm trap: unreachable
wasm backtrace:
    0: 0xa5d8 - runtime.abort
                    at /Users/dkegel/src/tinygo/src/runtime/runtime_tinygowasm.go:63:6
              - runtime._panic
                    at /Users/dkegel/src/tinygo/src/runtime/panic.go:14:2
    1: 0x10a5c - (*internal/task.Task).Resume
                    at /Users/dkegel/src/tinygo/src/internal/task/task_coroutine.go:23:12
              - runtime.scheduler
                    at /Users/dkegel/src/tinygo/src/runtime/scheduler.go:171:11
              - runtime.run
                    at /Users/dkegel/src/tinygo/src/runtime/scheduler_any.go:24:11
              - _start
                    at /Users/dkegel/src/tinygo/src/runtime/runtime_wasm_wasi.go:21:5
```

The panic goes away with both 0.19 and dev if I add the line
```
fmt.Fprintf(os.Stdout, "")
```
before a 12-parameter call to fmt.Fprintf, or break up the big call into two smaller ones.

aykevl, 2021-08-03:
Hmm, this smells like some sort of memory corruption. Can you try this patch to be sure? I suspect it will print something weird:

```diff
diff --git a/src/reflect/type.go b/src/reflect/type.go
index 2e709675..3d203712 100644
--- a/src/reflect/type.go
+++ b/src/reflect/type.go
@@ -608,6 +608,7 @@ func (t rawType) Align() int {
        case Array:
                return t.elem().Align()
        default:
+               println("type:", t.Kind().String())
                panic("unimplemented: alignment of type")
        }
 }
```

> tinygo build -target=wasi -scheduler=coroutines -gc=leaking -wasm-abi=generic

Also, note that you can remove `-scheduler=coroutines` and `-wasm-abi=generic` (they're already the default for `-target=wasi`).

dkegel-fastly, 2021-08-03:
As you suspected, with that patch on the tip of dev, it prints

type: invalid

(and on this run, I also saw a couple "panic: (reflect.Value).Interface: unexported")
so it does seem like memory corruption. 

Thanks for the tip re defaults.  I am still using an older tinygo, so I may leave the options there for a while.

deadprogram, 2021-11-10:
How is this issue with the latest `dev` branch? Same errors, or some different ones now?

dgryski, 2021-11-10:
My guess is this was memory corruption from the coroutine scheduler but I'l let @dkegel-fastly chime in if he's still been experiencing this on asyncify.

dkegel-fastly, 2021-11-10:
I haven't been seeing it with asyncify, but then, I haven't tried poking the bear by removing workarounds.

dkegel-fastly, 2021-12-15:
The large printf no longer crashes; tested with current dev, building app with 
-target=wasi -scheduler=asyncify -gc=conservative -wasm-abi=generic

closed 2021-12-15 by dkegel-fastly