tinygo-org/tinygo #2024: Wifi support for ESP32/ESP8266?

Technerder, 2021-08-02:
I recently stumbled upon this project and was thrilled to see that some basic features such as I2C and SPI were supported on the ESP32/ESP8266. Are there any plans to support Wifi on the chips and if so when can we expect to see a bare/basic implementation in the main code base?

gedw99, 2021-08-03:
i am also hoping for this... :)

aykevl, 2021-08-03:
I would absolutely love to see WiFi/BT support on these chips! However, it's going to be difficult because it doesn't have a lot of documentation. I've ordered an ESP32C3 which I want to try to add WiFi support to. Specifically the ESP32C3 because it uses the well documented RISC-V architecture instead of the sparsely documented (if at all) Xtensa architecture in the original ESP32. So my idea is that it should be a lot easier for that reason and when that works, I can use that knowledge for the original ESP32.

gedw99, 2021-08-03:
New hardware news 

https://www.reddit.com/r/homeautomation/comments/owzpou/espressif_to_release_new_esp32h2_microcontroller/?utm_source=share&utm_medium=ios_app&utm_name=iossmf

Dorbmon, 2021-08-05:
I hope too...

scroot, 2021-08-15:
i am also hoping for this...

willmendil, 2021-10-02:
any update on this?

aykevl, 2021-10-02:
Yes, please follow #2138 and https://github.com/tinygo-org/drivers/pull/320.

l1f, 2022-08-23:
thank you @aykevl for your work! Are there any updates on this?

ngandalf, 2022-10-16:
is there any news about wifi support on esp8266 / esp32 ?

conotto, 2022-10-29:
You would be better off using ESP-AT firmware on your ESP and control it with something well supported like RP2040 (yes, i realize thats 2 chips but at least you can have something that works now).
ESP8266/ESP32/ESP32-S2 are no longer actively developed or supported by Espressif, they now only recommend using ESP32-C2/C3 chips.

This is what i ended up doing.

ericmigi, 2022-12-30:
@conotto I would love to chat about how you solved this - any chance you could DM me on twitter.com/ericmigi or send me an email (username@gmail.com)?

CrazyPandar, 2023-01-12:
I found an ESP wifi driver, maybe this helps. 

https://github.com/esp-rs/esp-wireless-drivers-3rdparty

aykevl, 2023-01-12:
@CrazyPandar yes that's what we should use (see https://github.com/tinygo-org/drivers/pull/320#issuecomment-1284080185).

DatanoiseTV, 2023-01-21:
In rust, this was implemented: https://github.com/esp-rs/esp-wifi
Maybe this could be adopted to TinyGo?

aykevl, 2023-01-22:
@DatanoiseTV yes, see my comment above (which links to https://github.com/tinygo-org/drivers/pull/320#issuecomment-1284080185). TL;DR: we should indeed use https://github.com/esp-rs/esp-wireless-drivers-3rdparty. But that's a lot of work and nobody has stepped up to do it yet.

pcting, 2024-02-25:
what would be the process of integrating the [esp-wireless-drivers-3rdparty](https://github.com/esp-rs/esp-wireless-drivers-3rdparty) code? i could give it a try, but i don't have much embedded experience. are we talking about linking in the `.a` files and building a set of functions that wrap functions exposed in the c header files through `cgo`?

aykevl, 2024-02-25:
@pcting please follow this PR instead: https://github.com/tinygo-org/drivers/pull/650
Once the ESP32-C3 is implemented, we could take a look at supporting the ESP32/ESP8266 but those are going to be a lot more work because they also need Xtensa interrupt support. I spent a few days trying to get Xtensa interrupt support working at some point, but gave up on it.

I recommend following the ESP32-C3 work.

mamoit, 2024-12-28:
There seems to have been a significant development on this topic:
https://media.ccc.de/v/38c3-liberating-wi-fi-on-the-esp32

aykevl, 2024-12-31:
@mamoit true, but it doesn't help us now. Many parts are still missing, so it's not something we can use right now.

(I was there in the room when it was presented, it was an interesting talk!)

dector, 2025-03-24:
Is it possible to donate to fund work on this issue somehow? It would be awesome.

Thanks for your awesome work btw.

deadprogram, 2025-03-24:
@dector yes, please, and thank you!

https://opencollective.com/tinygo

JunBys, 2025-07-06:
Excuse me, is there any progress on this problem?

aykevl, 2025-07-06:
> Yes, please follow [#2138](https://github.com/tinygo-org/tinygo/pull/2138) and [tinygo-org/drivers#320](https://github.com/tinygo-org/drivers/pull/320).

@JunBys please follow the above PRs. I'll post an update when I get to work on it again.

niemeyer, 2025-09-18:
> [@dector](https://github.com/dector) yes, please, and thank you!
> 
> https://opencollective.com/tinygo

Just became a monthly contributor as well. It's not much at all, but at least it's one more voice towards radio support in cheap and widely available boards, as that tends to be the deal breaker for many use cases.  ESP32 S/C and RP2040/2350 are probably some of the most interesting candidates in that regard, at the moment.

Thanks for your long term work in TinyGo.

aykevl, 2025-10-05:
Hi @officer47p thank you for your interest! You can perhaps message me in the Gophers Slack? Or message me in some other way: https://aykevl.nl/about
Then I can get you started with wifi support, at least as far as I got last time.

davecheney, 2026-10-28:
The ESP8266 Wi-Fi request is already tracked in #1427, so I’m closing this as a duplicate to keep that discussion in one place. ESP32-C3 and ESP32-S3 Wi-Fi support are separate efforts and aren’t covered by this duplicate. See #1427 for ESP8266 updates.

closed 2026-10-28 by davecheney