tinygo-org/tinygo #2025: [GameBoyAdvance] Tinygo build don't build a correct GBA ROM and execute .gba file with mgba fails: "The game crashed!"

scraly, 2021-08-04:
Hi,

I tested to run and build the gba-display example found in this repository.

Cool, the tinygo run command is orking:

```
$ tinygo run -target=gameboy-advance gba-display.go
tinygo:ld.lld: warning: lld uses blx instruction, no object with architecture supporting feature detected
```
[image]

But unfortunately I can't build successfully the gba binary and when I test it through mgba emulator, it fails! :( 

```
$ tinygo build -target=gameboy-advance -o bin/gba-display.gba gba-display.go 
tinygo:ld.lld: warning: lld uses blx instruction, no object with architecture supporting feature detected

$ file bin/gba-display.gba
bin/gba-display.gba: data

$ mgba bin/gba-display.gba
The game crashed!
```

So, I can't build a working gba ROEM file.

Is it possible to know how to remove the warning message and to fix the bug?

Thanks

deadprogram, 2023-05-22:
This is no longer a problem at least with the 0.28 release coming soon. Tagging to close this issue on next release. Thanks @scraly

scraly, 2023-05-22:
Thanks @deadprogram 
I will test it soon! :)