tinygo-org/tinygo #2026: [Suggestion] aarch64 deb

Hero9909, 2021-08-05:
it would be awesome if a build for aarch64 could be provided as the current release only ships armhf.

My usecase would be a easy installation in termux as i would love to run a few things from my phone while traveling by train.

about limitations i remember that i did a espressif install about a year ago and did a flash that acually worked. 
i appreciate any corrections if im wrong.

LaPingvino, 2021-09-27:
Also useful for many ChromeOS systems, as they come with Debian aarch64 on ARM systems.

cancecen, 2022-05-31:
Any plans of doing this? @deadprogram I noticed that this was marked as an enhancement, wondering if you have any plans for doing this soon.

aykevl, 2022-05-31:
We have support for this in the dev branch, the next release will have an aarch64 binary: https://github.com/tinygo-org/tinygo/pull/2835

cancecen, 2022-05-31:
@aykevl this is great thank you. Any rough idea on when next release would be?

aykevl, 2022-05-31:
No plans currently, but maybe it would be a good idea in the not too distant future.

deadprogram, 2022-06-01:
We will have a release soonish.

closed 2022-07-01 by deadprogram