tinygo-org/tinygo #2027: "tinygo test testing" doesn't find any source files?

dkegel-fastly, 2021-08-05:
This works:
```go test testing```

This also works:
```tinygo test testing/iotest```

But for some reason, 
```tinygo test testing```
fails with
``` 
?   	testing	[no test files]
```

peterbourgon, 2021-08-09:
`go test` takes as an argument the package(s) you intend to test. By passing `testing` you're expressing the stdlib `pacakge testing` which in normal Go contains `_test.go` files and in TinyGo does not. So this is expected behavior, but the more interesting question is: why are you doing this? Testing `package testing` itself is something the Go (or TinyGo) authors should do, but I'm not sure why you'd want to?

aykevl, 2021-08-09:
Apparently this already works: https://github.com/tinygo-org/tinygo/pull/2037
I suspect you needed to run `tinygo clean` to get it to work. Just like the syscall package, the cached GOROOT doesn't immediately reflect changes in the ./src/testing directory.

> By passing `testing` you're expressing the stdlib `pacakge testing` which in normal Go contains `_test.go` files and in TinyGo does not.

See: https://github.com/tinygo-org/tinygo/pull/1814#discussion_r683824257

> but the more interesting question is: why are you doing this? Testing `package testing` itself is something the Go (or TinyGo) authors should do, but I'm not sure why you'd want to?

Exactly. We would like to do this in CI. It would be useful to test this package, just like any other package.

peterbourgon, 2021-08-09:
Ah, my apologies, I was missing context, which #2037 provides. Looks good.

dkegel-fastly, 2021-08-09:
Thanks for following up on this.  I was simply confused about the mechanics.

closed 2021-08-09 by dkegel-fastly