tinygo-org/tinygo #2028: TinyGo Wasm keeps freezing or stop working on processing thousands of data

fenezema, 2021-08-06:
Hello,
Currently, I'm facing this issue when loading data to memory.

I load data from browser localStorage. And because of the nature of browser localStorage can only store string data, there are a lot of repeated process of decode these data and map these data to struct. At some point, during those loop, I will find 2 - 3 data taking for about 3 - 5 seconds each to finish while the others only take less than 20ms to map. And no error appears on the browser console as well during this process.

The number of data I load is about 2000-3000 data. But it's not a problem if the number of data only around 100-200 data.
And it seems that it works just fine if I compile this with Go regardless the number of data I use.

I've been trying to figure things out for some days now, and has not been able to solve this problem.
I also have no clue if this is a memory issue or something else.
I also did try to solve this by not loading the data to memory, but it seems it's still happening for some reason.

Any information regarding this issue is very much appreciated.
Thank you

TinyGo version:
`tinygo version 0.18.0 linux/amd64 (using go version go1.16.5 and LLVM version 11.0.0)`

codefromthecrypt, 2022-09-07:
@fenezema it is hard to determine what the issue is except possibly garbage collection backup. If you are past this issue, please close this issue. Otherwise, it would be easier to troubleshoot if you had a Github repository to share which can reproduce the problem.

aykevl, 2022-09-15:
I suspect this is a performance cliff in the garbage collector. I tried to fix it in #2880 but it doesn't work as expected. I still need to figure out why it fails.

dgryski, 2022-09-15:
I agree it's probably related to the garbage collector.  Do you have some sample code we could play with?

dgryski, 2023-02-02:
Closing for now.  There have been improvements in the garbage collector. If you're still encountering errors, please provide a test case for us to debug with.

closed 2023-02-02 by dgryski