tinygo-org/tinygo #2036: [esp32] SPI not working as expected

Hero9909, 2021-08-07:
my setup:
esp32 [link](https://www.az-delivery.de/en/products/esp32-developmentboard)
waveshare 1.5 inch oled display [link](https://www.waveshare.com/1.5inch-oled-module.htm)
goland ide with tinygo plugin
tinygo 0.19.0

tl;dr;
display is working with esp, arduino spi and arduino ide no additional config, but not with default tinygo esp32 spi. soft spi with tinygo is working. docs missread? implementation wrong? board not expected for implementation?

my story in all the details of the last 24h:

ive tried to get a simple program working to control a 1.5 inch oled display from waveshare.

for that ive tried this first using arduino and its spi library.

```
#include <SPI.h>
void setup() {
  pinMode(17, OUTPUT); // CS
  pinMode(16, OUTPUT); // RST
  pinMode(4, OUTPUT); // DC
  SPI.begin();
  digitalWrite(16,HIGH);
  delay(100);
  digitalWrite(16,LOW);
  delay(100);
  digitalWrite(16,HIGH);
  delay(100);
  SPI.transfer(0xaf);
  delay(300);
  SPI.transfer(0xa5);
}
void loop() {}
```

that works as intended and resets enables and fills the display with white pixels.

but trying that with tinygo doesnt work as intended, 

```
package main

import (
	"machine"
	"time"
)

func main() {
	dc := machine.Pin(4)
	dc.Configure(machine.PinConfig{Mode: machine.PinOutput})
	rst := machine.Pin(16)
	rst.Configure(machine.PinConfig{Mode: machine.PinOutput})
	cs := machine.Pin(17)
	cs.Configure(machine.PinConfig{Mode: machine.PinOutput})
	dc.Low()
	cs.Low()
	rst.High()
	time.Sleep(100 * time.Millisecond)
	rst.Low()
	time.Sleep(100 * time.Millisecond)
	rst.High()
	time.Sleep(100 * time.Millisecond)
	machine.SPI2.Transfer(0xaf)
	time.Sleep(300 * time.Millisecond)
	machine.SPI2.Transfer(0xa5)
}
```
ive tried to find the difference between those two implementations in the spi protcol. ive done all testing with the same wiring.
ive experimented with all mentioned frequencies and modes, no good results. so ive decided to build a very slow software spi emulation at the bare metal of the protocol:

```
package main

import (
	"machine"
	"time"
)

func main() {
	dc := machine.Pin(4)
	dc.Configure(machine.PinConfig{Mode: machine.PinOutput})
	rst := machine.Pin(16)
	rst.Configure(machine.PinConfig{Mode: machine.PinOutput})
	cs := machine.Pin(17)
	cs.Configure(machine.PinConfig{Mode: machine.PinOutput})
	do := machine.Pin(23)
	do.Configure(machine.PinConfig{Mode: machine.PinOutput})
	clk := machine.Pin(18)
	clk.Configure(machine.PinConfig{Mode: machine.PinOutput})

	dc.Low()
	cs.High()
	
	rst.High()
	time.Sleep(100 * time.Millisecond)
	rst.Low()
	time.Sleep(100 * time.Millisecond)
	rst.High()
	time.Sleep(100 * time.Millisecond)

	cs.Low()
	do.Low()
	for i := 7; i >= 0; i-- {
		do.Set(((0xaf>>i)&0x1)>0)
		clk.High()
		time.Sleep(2 * time.Millisecond)
		clk.Low()
	}
	time.Sleep(300 * time.Millisecond)
	for i := 7; i >= 0; i-- {
		do.Set(((0xa5>>i)&0x1)>0)
		clk.High()
		time.Sleep(2 * time.Millisecond)
		clk.Low()
	}
	cs.High()
}
```

and its actually working even if im not proud of this pile of ...
but that brings me to this bug. ive proved that the wiring is working in some way and tinygo is captable of communicating with the display, but not with the current configuration or implementation as i cant exclude that ive missed something in the docs.

lastly can someone test this hypothesis before i blame myself?

sago35, 2021-09-16:
@Hero9909 

Thank you for creating the issue.

The same thing happened to me when I tried to run ili9341.
I think it doesn't work for some devices because SCK and SDO work almost simultaneously.

Perhaps Frequency from 20e6 to 40e6-1 works fine?

I could probably fix it, and I'll try to sort it out and create a PR.

dev branch - 3eb9dca695254b9d59f30597ea6c409e1b65e7ee
[image]

new version
[image]

deadprogram, 2021-11-18:
Closing as this was fixed by #2148 thank you!

closed 2021-11-18 by deadprogram