tinygo-org/tinygo #2045: How to export memory or a global variable?

zu1k, 2021-08-14:
I want to expose a memory or variable, my code is like that:

```
//export id
var id = "id_007"

//export const_str
const const_str = "const str"
```

But I found that these variables or constants were not being exported, maybe, I can't found them.

How can I export a memory, a variable correctly? Thanks!

vlkv, 2022-06-02:
I am also curious how to do that. Can somebody answer or throw an url to the docs?..

radkomih, 2022-07-01:
Looks like exporting `globals, memory` is not currently supported?
I hope someone to confirm that.

radkomih, 2022-07-01:
Hmm, I've asked because I have `wasm` compiled from `rust` that looks like the one below. I am not sure how these `Globals` got exported?

```js
Type: wasm
Imports:
  Functions:
    "env"."Another_provider_func": [I32, I64, I64] -> []
    ...
  Memories:
    "env"."memory": not shared (20 pages..)
  Tables:
  Globals:
Exports:
  Functions:
    "My_custom_func": [I32, I32] -> [I64]
  Memories:
  Tables:
    "__indirect_function_table": FuncRef (352..352)
  Globals:
    "__data_end": I32 (constant)
    "__heap_base": I32 (constant)
```

aykevl, 2022-07-01:
You mean `__data_end` and `__heap_base`? Those are defined by the linker, not by Rust. I'm not sure how they were exported.

closed 2023-10-11 by zu1k