tinygo-org/tinygo #2046: Cannot use len(s) (value of type int) as uintptr value in assignment

StarpTech, 2021-08-14:
Hi, I see the following error when compiling a program with tinygo. Is it unsupported by design?
```
❯ tinygo build -o hello ./main.go
# github.com/cespare/xxhash
../../../go/pkg/mod/github.com/cespare/xxhash@v1.1.0/xxhash_unsafe.go:27:11: cannot use len(s) (value of type int) as uintptr value in assignment
../../../go/pkg/mod/github.com/cespare/xxhash@v1.1.0/xxhash_unsafe.go:28:11: cannot use len(s) (value of type int) as uintptr value in assignment
```
Source: https://github.com/cespare/xxhash/blob/569f7c8abf1f58d9043ab804d364483cb1c853b6/xxhash_unsafe.go#L27

I also saw similar issues in https://github.com/tidwall/gjson/blob/5c2e4b382486589dad7478130a364ee2fa6a068b/gjson_ngae.go#L74

aykevl, 2021-08-14:
This is a duplicate of #1284.

In short, the code is not portable. However, in this case it's easy to work around it. This slightly changed version also compiles under TinyGo:

```go
func Sum64String(s string) uint64 {
	// See https://groups.google.com/d/msg/golang-nuts/dcjzJy-bSpw/tcZYBzQqAQAJ
	// for some discussion about this unsafe conversion.
	var b []byte
	bh := (*reflect.SliceHeader)(unsafe.Pointer(&b))
	sh := (*reflect.StringHeader)(unsafe.Pointer(&s))
	bh.Data = sh.Data
	bh.Len = sh.Len
	bh.Cap = sh.Len
	return Sum64(b)
}
```

aykevl, 2021-08-14:
Closing because it is a duplicate.

closed 2021-08-14 by aykevl