tinygo-org/tinygo #2047: wasm: -opt=z output is larger than -opt=s or -opt=2

jfcg, 2021-08-14:
Hi,
On Manjaro Linux with core i5 64-bit x86_64 CPU and tinygo `0.19.0`. main.go:
```go
package main

import "fmt"

func main() {
	fmt.Println("Hello, WebAssembly!")
}
```
when compiled with
```
tinygo build -opt=z -no-debug -target wasm -o main.wasm main.go
```
output file is `71645` bytes. But `-opt=s` or `-opt=2` both yield an output of `37612` bytes, quite a difference. As per [options doc](https://tinygo.org/docs/reference/usage/important-options/) `-opt=z` should yield the smallest file.

aykevl, 2021-08-14:
Yes, this can sometimes happen.

Typically, `-opt=z` should result in the smallest binary size. However, compiler optimizations are often heuristics which can go wrong. In this case, it's likely a function is inlined with `-opt=s` and resulted in a further code size reduction after inlining. `-opt=z` is less likely to inline, which is usually a good thing for code size, but it can sometimes have a big difference.

I don't think this is a bug, just a side effect of how the optimizer works.

deadprogram, 2021-09-07:
Closing, since the question was answered. Please reopen if needed. Thanks!

closed 2021-09-07 by deadprogram