tinygo-org/tinygo #2053: Docker: cannot update packages inside container with "apt update"

Supportic, 2021-08-16:
This output is what I get. Seems like some debian repos are changed.

```shell
root@f6aff3bf815e:/# apt update
Ign:1 http://security.debian.org/debian-security stable/updates InRelease
Err:2 http://security.debian.org/debian-security stable/updates Release
  404  Not Found [IP: 151.101.66.132 80]
Get:3 http://deb.debian.org/debian stable InRelease [113 kB]
Get:4 http://deb.debian.org/debian stable-updates InRelease [36.8 kB]
N: Repository 'http://deb.debian.org/debian stable InRelease' changed its 'Version' value from '10.10' to '11.0'
E: Repository 'http://deb.debian.org/debian stable InRelease' changed its 'Codename' value from 'buster' to 'bullseye'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
Do you want to accept these changes and continue updating from this repository? [y/N] N
N: Repository 'http://deb.debian.org/debian stable-updates InRelease' changed its 'Version' value from '' to '11-updates'
E: Repository 'http://deb.debian.org/debian stable-updates InRelease' changed its 'Codename' value from 'buster-updates' to 'bullseye-updates'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
Do you want to accept these changes and continue updating from this repository? [y/N] N
Reading package lists... Done
E: The repository 'http://security.debian.org/debian-security stable/updates Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

aykevl, 2021-08-17:
We probably need to update the Go version of the container.

See also: https://github.com/tinygo-org/tinygo/pull/2051

Supportic, 2021-08-17:
@aykevl does the latest golang image not contain the latest go version?
https://github.com/tinygo-org/tinygo/blob/release/Dockerfile

deadprogram, 2021-08-18:
It contains 1.16 which was the current version until this week.

The problem however has to do with some other dependencies in the container from LLVM needing to be updated, I think. Same issue with the github action to build the dockerfile from the `dev` branch.

deadprogram, 2021-08-18:
@Supportic I think that this PR https://github.com/tinygo-org/tinygo/pull/2058 will probably fix your problem, along with the problem in the docker automatic build from the `dev` branch.

deadprogram, 2021-08-19:
The `dev` branch now contains corrected Dockerfile and GH action is passing.

Supportic, 2021-08-19:
Hey, thanks for the update. Trying to build the Dockerfile in the dev branch but it's giving me an error:
```shell
executor failed running [/bin/sh -c wget -O- https://apt.llvm.org/llvm-snapshot.gpg.key| apt-key add - &&     echo "deb http://apt.llvm.org/buster/ llvm-toolchain-buster-11 main" >> /etc/apt/sources.list &&     apt-get update &&     apt-get install -y llvm-11-dev libclang-11-dev lld-11 git]: exit code: 100
```
With the hint:
`Resolving apt.llvm.org (apt.llvm.org)... Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).`

deadprogram, 2021-08-19:
That latter is just a warning not an error. You did not seem to including any actually error in the output?

Supportic, 2021-08-19:
@deadprogram This is the whole console output:
```shell
D:\Desktop\tinygo>docker build .
[+] Building 5.6s (6/18)
 => [internal] load build definition from Dockerfile                                                               0.0s
 => => transferring dockerfile: 32B                                                                                0.0s
 => [internal] load .dockerignore                                                                                  0.0s
 => => transferring context: 2B                                                                                    0.0s
 => [internal] load metadata for docker.io/library/golang:1.16                                                     0.9s
 => CACHED [tinygo-base 1/6] FROM docker.io/library/golang:1.16@sha256:87cbbe43ece5024f0745be543c81ae6bf7b88291a8  0.0s
 => [internal] load build context                                                                                  0.1s
 => => transferring context: 48.36kB                                                                               0.1s
 => ERROR [tinygo-base 2/6] RUN wget -O- https://apt.llvm.org/llvm-snapshot.gpg.key| apt-key add - &&     echo "d  4.5s
------
 > [tinygo-base 2/6] RUN wget -O- https://apt.llvm.org/llvm-snapshot.gpg.key| apt-key add - &&     echo "deb http://apt.llvm.org/buster/ llvm-toolchain-buster-11 main" >> /etc/apt/sources.list &&     apt-get update &&     apt-get install -y llvm-11-dev libclang-11-dev lld-11 git:
#5 0.352 --2021-08-19 15:50:50--  https://apt.llvm.org/llvm-snapshot.gpg.key
#5 0.362 Resolving apt.llvm.org (apt.llvm.org)... Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
#5 0.792 199.232.194.49, 199.232.198.49, 2a04:4e42:4c::561, ...
#5 0.792 Connecting to apt.llvm.org (apt.llvm.org)|199.232.194.49|:443... connected.
#5 0.887 HTTP request sent, awaiting response... 200 OK
#5 0.922 Length: 3145 (3.1K) [application/octet-stream]
#5 0.922 Saving to: 'STDOUT'
#5 0.923
#5 0.923      0K ...                                                   100% 10.8M=0s
#5 0.923
#5 0.923 2021-08-19 15:50:51 (10.8 MB/s) - written to stdout [3145/3145]
#5 0.923
#5 1.330 OK
#5 1.538 Get:1 http://deb.debian.org/debian bullseye InRelease [113 kB]
#5 1.542 Get:2 http://security.debian.org/debian-security bullseye-security InRelease [44.1 kB]
#5 1.577 Get:3 http://deb.debian.org/debian bullseye-updates InRelease [36.8 kB]
#5 1.675 Get:5 http://security.debian.org/debian-security bullseye-security/main amd64 Packages [25.8 kB]
#5 1.779 Get:6 http://deb.debian.org/debian bullseye/main amd64 Packages [8178 kB]
#5 1.916 Get:4 https://apt.llvm.org/buster llvm-toolchain-buster-11 InRelease [5525 B]
#5 2.119 Get:7 https://apt.llvm.org/buster llvm-toolchain-buster-11/main amd64 Packages [14.7 kB]
#5 3.193 Fetched 8417 kB in 2s (4588 kB/s)
#5 3.193 Reading package lists...
#5 3.656 Reading package lists...
#5 4.101 Building dependency tree...
#5 4.219 Reading state information...
#5 4.290 git is already the newest version (1:2.30.2-1).
#5 4.290 Some packages could not be installed. This may mean that you have
#5 4.290 requested an impossible situation or if you are using the unstable
#5 4.290 distribution that some required packages have not yet been created
#5 4.290 or been moved out of Incoming.
#5 4.290 The following information may help to resolve the situation:
#5 4.290
#5 4.290 The following packages have unmet dependencies:
#5 4.371  libclang-11-dev : Depends: libstdc++-8-dev but it is not installable
#5 4.371                    Depends: libgcc-8-dev but it is not installable
#5 4.371                    Depends: libobjc-8-dev but it is not installable
#5 4.371                    Depends: libclang-common-11-dev (= 1:11.1.0~++20210805102431+1fdec59bffc1-1~exp1~20210805203107.165) but it is not going to be installed
#5 4.372  libllvm11 : Depends: libffi6 (>= 3.0.4) but it is not installable
#5 4.385 E: Unable to correct problems, you have held broken packages.
------
executor failed running [/bin/sh -c wget -O- https://apt.llvm.org/llvm-snapshot.gpg.key| apt-key add - &&     echo "deb http://apt.llvm.org/buster/ llvm-toolchain-buster-11 main" >> /etc/apt/sources.list &&     apt-get update &&     apt-get install -y llvm-11-dev libclang-11-dev lld-11 git]: exit code: 100
```

Supportic, 2021-08-19:
Sorry for the inconvenience. The problem was not in the code, the problem was sitting infront of the machine.
Pulling the repo is one thing but not checking out the dev branch is the other thing...
From the output you can read that is was still pointing to: http://apt.llvm.org/buster/

After checking out the dev branch and building it successfully the `apt update` command worked inside the container.

Thanks

closed 2021-08-19 by Supportic