tinygo-org/tinygo #2054: tinygo problems blocking support for newly released go 1.17?

dkegel-fastly, 2021-08-17:
Lucky me, I happened to install the latest go right when go 1.17 was released, and discovered A Problem Or Two :-) running tinygo.
````
$ go version
go version go1.17 darwin/amd64
$ tinygo version
tinygo version 0.20.0-dev darwin/amd64 (using go version go1.17 and LLVM version 11.1.0)
````

First problem:

```
$ tinygo build -o foo
error: could not read version from GOROOT (/usr/local/go): Invalid go version output:
```

dtruss shows
```
 open("/usr/local/go/src/runtime/internal/sys/zversion.go\0", 0x1000000, 0x0)		 = 3 0
 read(0x3, "// Code generated by go tool dist; DO NOT EDIT.\n\npackage sys\n\nconst StackGuardMultiplierDefault = 1\n\0", 0x200)		 = 100 0
 write(0x2, "error: could not read version from GOROOT (/usr/local/go): Invalid go version ...
```

Looks like https://github.com/golang/go/commit/5159c836410535 removed the line ```const TheVersion = `%s``` that tinygo's GorootVersionString() was expecting.

That's an easy fix - make the first hunk fall through if zversion.go doesn't match the regexp.

Once past that, the next problem is:

```
$ tinygo build -o foo
error: requires go version 1.15 through 1.16, got go1.17
```

tinygo's builder/config.go says
```
	if major != 1 || minor < 15 || minor > 16 {
		return nil, fmt.Errorf("requires go version 1.15 through 1.16, got go%d.%d", major, minor)
```

Once past that, tinygo's own test suite fails as follows (lightly summarized):
```
$ go test
Undefined symbols for architecture x86_64:
  "_math.archFloor", referenced from:
      _runtime.run$1$gowrapper in main.o
  "_math.archHypot", referenced from:
      _runtime.run$1$gowrapper in main.o
  "_math.archMax", referenced from:
      _runtime.run$1$gowrapper in main.o
  "_math.archCeil", referenced from:
      _runtime.run$1$gowrapper in main.o
  "_math.archMin", referenced from:
      _runtime.run$1$gowrapper in main.o
  "_math.archTrunc", referenced from:
      _runtime.run$1$gowrapper in main.o
  "_math.archExp", referenced from:
      _math.Exp in main.o
  "_math.archLog", referenced from:
      _math.Log in main.o
...
ld.lld: error: undefined symbol: math.archSqrt
>>> referenced by sqrt.go:94 (/usr/local/go/src/math/sqrt.go:94)
>>>               /var/folders/v0/0k9hwftd29b18z9z7s2g2sfc0000gn/T/tinygo168732285/main.o:(.text.math.Sqrt+0x0)
...
--- FAIL: TestCompiler (0.00s)
        --- FAIL: TestCompiler/EmulatedCortexM3/json.go (13.25s)
            main.go:931: # encoding/json
            main.go:901: /usr/local/go/src/encoding/json/encode.go:1249:13: sf.IsExported undefined (type reflect.StructField has no field or method IsExported)
    --- FAIL: TestCompiler/Host (0.00s)
        --- FAIL: TestCompiler/Host/json.go (18.51s)
            main.go:931: # encoding/json
            main.go:901: /usr/local/go/src/encoding/json/encode.go:1249:13: sf.IsExported undefined (type reflect.StructField has no field or method IsExported)
        --- FAIL: TestCompiler/Host/filesystem.go (10.11s)
            main.go:931: # io/ioutil
            main.go:901: /usr/local/go/src/io/ioutil/tempfile.go:40:12: MkdirTemp not declared by package os
```

at which point I decided it was time to walk away slowly from go 1.17 for now.

More detailed system info: macOS Big Sur 11.5.2, with system llvm-gcc and fresh-ish upstream clang:
```
$ llvm-gcc --version
Apple clang version 12.0.5 (clang-1205.0.22.11)
Target: x86_64-apple-darwin20.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
$ clang --version
clang version 11.1.0
Target: x86_64-apple-darwin20.6.0
Thread model: posix
InstalledDir: /usr/local/bin
```

Nerzal, 2021-08-17:
This is done on purpose. Sometimes Go introduces changes, that need to be tested, before TinyGo adapts the support. 
This is why a max Go version is defined in each TinyGo release. 

This results in TinyGo taking a little longer to adapt to newer Go releases, but also results in higher quality and less problems with Tinygo.

dkegel-fastly, 2021-08-17:
Ha, Ayke is way ahead of the game, #2048 will fix this.

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram