tinygo-org/tinygo #2057: TinyGo   strconv.FormatInt()   unreachable

AdwindOne, 2021-08-18:
strconv.FormatInt()   function convert bigint(eg over one billion)  to string  happen to  unreachable

aykevl, 2021-09-01:
Can you please add the code that will reproduce this error?

deadprogram, 2021-09-07:
@AdwindOne can you please provide a code sample we can use to reproduce? Thank you.

dgryski, 2023-02-02:
Not reproducible.
```
~/go/src/github.com/dgryski/bug $ go run bug.go
4000000000
~/go/src/github.com/dgryski/bug $ tinygo run bug.go
4000000000
~/go/src/github.com/dgryski/bug $ tinygo run -target=wasi bug.go
4000000000
~/go/src/github.com/dgryski/bug $ cat bug.go
package main

import (
	"fmt"
	"strconv"
)

func main() {

	var i64 int64 = 4_000_000_000

	fmt.Println(strconv.FormatInt(i64, 10))
}
```