tinygo-org/tinygo #2059: Build is broken (unable to get go version)

andrewmed, 2021-08-18:
This commit https://github.com/tinygo-org/tinygo/pull/1705 introduced parsing physical path in `src/runtime/internal/sys/zversion.go` in go toolchain to get go version

However line with  was recently removed from go compiler in https://github.com/golang/go/commit/5159c836410535fad0ae0b23329ef91347c1b133#diff-b769e614adb0e7a4c2eb1e24f92df350ddb3afe2a3253293af8279afc60eb3d9L31

Therefore tinygo buils are broken

```
andy@desktop frontend % ~/code/tinygo/tinygo build -o main.wasm -target wasm main.go
error: could not read version from GOROOT (/Users/andy/code/goroot): Invalid go version output:
// Code generated by go tool dist; DO NOT EDIT.

package sys

const StackGuardMultiplierDefault = 1

andy@desktop frontend % ls /Users/andy/code/goroot/src/runtime/internal/sys/zversion.go 
/Users/andy/code/goroot/src/runtime/internal/sys/zversion.go
andy@desktop frontend % cat /Users/andy/code/goroot/src/runtime/internal/sys/zversion.go
// Code generated by go tool dist; DO NOT EDIT.

package sys

const StackGuardMultiplierDefault = 1
```

I would have submitted PR to simply use VERSION file from the go toolchain root, but there must be some reason it is not being used in first place, which I do not quite understand...

dkegel-fastly, 2021-08-18:
Dup of https://github.com/tinygo-org/tinygo/issues/2054

Try backing down to go 1.16.x for now.  Ayke's fix is ready and waiting on CI vendor to add go 1.17 image.

andrewmed, 2021-08-18:
> Dup of #2054
> 
> Try backing down to go 1.16.x for now. Ayke's fix is ready and waiting on CI vendor to add go 1.17 image.

OK
I was too fast to make PR too)

deadprogram, 2021-09-01:
Closing now that #2048 has been merged into `dev`. Please reopen if needed.

closed 2021-09-01 by deadprogram