tinygo-org/tinygo #2079: duplicate type names with different fields in different functions cause compilation failure

dgryski, 2021-09-03:
```
package main

func foo() {
	type kv struct {
		v int
	}

	data := []kv{
		{0}, {1}, {2},
	}

	for _, elt := range data {
		println(elt.v)
	}
}

func bar() {

	type kv struct {
		v string
	}

	data := []kv{
		{"v0"}, {"v1"}, {"v2"},
	}

	for _, elt := range data {
		println(elt.v)
	}

}

func main() {
	foo()
	bar()
}
```

Using `tinygo version 0.19.0 darwin/amd64 (using go version go1.16.7 and LLVM version 11.0.0)` produces

```
Stored value type does not match pointer operand type!
  store %runtime._string { i8* getelementptr inbounds ([2 x i8], [2 x i8]* @"main.bar$string", i32 0, i32 0), i64 2 }, i64* %2, align 8, !dbg !56
 i64Stored value type does not match pointer operand type!
  store %runtime._string { i8* getelementptr inbounds ([2 x i8], [2 x i8]* @"main.bar$string.1", i32 0, i32 0), i64 2 }, i64* %9, align 8, !dbg !60
 i64Stored value type does not match pointer operand type!
  store %runtime._string { i8* getelementptr inbounds ([2 x i8], [2 x i8]* @"main.bar$string.2", i32 0, i32 0), i64 2 }, i64* %12, align 8, !dbg !61
 i64Incorrect number of arguments passed to called function!
  call void @runtime.printstring(i64 %21, i8* undef, i8* null), !dbg !64
error: verification error after compiling package github.com/dgryski/bug
```

fgsch, 2021-09-03:
Also happens with dev as of `tinygo version 0.20.0-dev-c4d3eeab darwin/amd64 (using go version go1.17 and LLVM version 11.0.0)`.

aykevl, 2021-09-08:
Huh, that's an interesting bug. The cause is that named struct types are identified by their name and package. This is usually fine, for types declared at the package level. But inside functions, there can be duplicates as you just found out.

I have a fix here: https://github.com/tinygo-org/tinygo/pull/2094

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram