tinygo-org/tinygo #2080: Invalid Go version

Ulset, 2021-09-03:
On running 
`tinygo flash -target arduino Arduino.go`

I get the following error message:
`error: could not read version from GOROOT (/opt/homebrew/Cellar/go/1.17/libexec): Invalid go version output:`

dgryski, 2021-09-03:
This is https://github.com/tinygo-org/tinygo/pull/2048 that will be fixed in the next release.

deadprogram, 2021-09-03:
TinyGo 0.19.0 does not support Go 1.17.  Please try the `dev` branch if you need that support right away, downgrade to Go 1.16.x or just wait for the upcoming next release.

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram

dosgo, 2021-09-22:
0.20.0 No windows-amd64.zip package,does it support windows?

sago35, 2021-09-22:
@dosgo 
The binary for windows has been completed and is attached.
https://github.com/tinygo-org/tinygo/releases/tag/v0.20.0