tinygo-org/tinygo #2090: interp timeout compiling large maps with runtime inits

dgryski, 2021-09-06:
```
package main

import "strconv"

var x = map[string]int{}

func init() {
	x := map[string]int{}
	for i := 0; i < 65536; i++ {
		x[strconv.Itoa(i)] = i
	}
}

func main() {
	println(len(x))
}
```

Compilation fails with
```
/Users/dgryski/go/src/tinygo.org/tinygo/src/runtime/func.go:26:6: interp: running for more than a minute, timing out (executed calls: 182712)
  %val = alloca %runtime._string, align 8, !dbg !447

traceback:
/Users/dgryski/go/src/tinygo.org/tinygo/src/runtime/func.go:26:6:
  %val = alloca %runtime._string, align 8, !dbg !447
/Users/dgryski/go/src/tinygo.org/tinygo/src/runtime/hashmap.go:134:16:
  %69 = call i32 @runtime.getFuncPtr(i8* %keyEqual.context, i32 %keyEqual.funcptr, i8* @"reflect/types.funcid:func:{basic:unsafeptr,basic:unsafeptr,basic:uintptr}{basic:bool}", i8* undef, i8* null), !dbg !604
/Users/dgryski/go/src/tinygo.org/tinygo/src/runtime/hashmap.go:337:12:
  call void @runtime.hashmapSet(%runtime.hashmap* %m, i8* %key, i8* %value, i32 %3, i8* undef, i32 ptrtoint (%runtime._interface* @"runtime.hashmapStringEqual$withSignature" to i32), i8* undef, i8* undef), !dbg !458
main/<init>:
  call void @runtime.hashmapStringSet(%runtime.hashmap* %3, i8* %14336, i32 %14337, i8* bitcast (i32* @"main$alloca" to i8*), i8* undef, i8* null), !dbg !439
```

dgryski, 2021-09-08:
Would adding `//go:tinygo-init-at-runtime` or something be a reasonable work-around to move some globals to initialization at runtime instead of at compile-time with the `interp` pass ?

dkegel-fastly, 2021-09-18:
It doesn't feel like a very exciting prospect, somehow.

dkegel-fastly, 2022-01-10:
This patch attempts to help the developer figure out what init function is being slow, so he can rework it:
```
--- a/interp/interpreter.go
+++ b/interp/interpreter.go
@@ -19,7 +19,7 @@ func (r *runner) run(fn *function, params []value, parentMem *memoryView, indent
 
        if time.Since(r.start) > time.Minute {
                // Running for more than a minute. This should never happen.
-               return nil, mem, r.errorAt(fn.blocks[0].instructions[0], fmt.Errorf("interp: running for more than a minute, timing out (executed calls: %d)", r.callsExecuted))
+               return nil, mem, r.errorAt(fn.blocks[0].instructions[0], fmt.Errorf("interp: %s running for more than a minute, timing out (executed calls: %d)", fn.name, r.callsExecuted))
```

And sometimes it does.  But other times the slow function is runtime.hashmapMake, which while interesting is not as helpful as it could be.  Is there a way to print information about the caller (or call stack)?

dkegel-fastly, 2022-01-10:
Oh, never mind, it's printed already, I was just having trouble seeing it.

dankegel, 2022-05-01:
This seems to affect "tinygo test strings".  Even raising the limit to 30 minutes didn't rescue it, it still says
```
tinygo/src/sync/mutex.go:17:7: interp: running for more than 1800 seconds, timing out (executed calls: 7786197)
```

dgryski, 2022-05-02:
Yeah, it's creating a large random buffer. Pretty sure this hits `bytes` too.

dkegel-fastly, 2022-05-04:
Could we mark all code inside _test.go files as runAtRuntime?

dgryski, 2022-08-20:
Not sure this counts as "fixed". More like the first workaround is available, but the underlying quadratic behaviour of the interp pass still needs to be dealt with.

dgryski, 2023-09-19:
`interp` has definitely improve some; `strings` and `bytes` now pass the interp phase while testing, and I don't just think it's a higher timeout limit.  I wonder if something else is triggering move-to-runtime calls for those packages.

dkegel-fastly, 2026-02-15:
https://github.com/tinygo-org/tinygo/issues/3653 may be related