tinygo-org/tinygo #2092: Teensy 4.0's machine.UART.handleInterrupt() does not work

sago35, 2021-09-07:
@ardnew 
The program in examples/echo does not work.
It seems that `UART.handleInterrupt()` is not working.

The UART transmission by fmt.Printf() seems to be working.

ardnew, 2021-09-07:
Just to confirm - which branch are you using?

ardnew, 2021-09-07:
Whatever the issue, feel free to assign to me and I can investigate when i get home from work this evening

sago35, 2021-09-07:
I used the latest from the dev branch to check.

```
$ tinygo flash --target teensy40 --size short ./src/examples/echo
   code    data     bss |   flash     ram
  20672      48  131392 |   20720  131440
Teensy Loader, Command Line, Version 2.2
Read "C:\Users\sago3\AppData\Local\Temp\tinygo222034227\main.hex": 20720 bytes, 1.0% usage
Waiting for Teensy device...
 (hint: press the reset button)
Found HalfKay Bootloader
Read "C:\Users\sago3\AppData\Local\Temp\tinygo222034227\main.hex": 20720 bytes, 1.0% usage
Programming...............
Booting

$ git rev-parse HEAD
eaab05fc43129d3ac816ece486498c0b752a1262
```

ardnew, 2021-09-11:
@sago35 Let me know if this fixes UART for you.

Also, let me know if you need any instructions to reconfigure that `feature/usb-common` branch to support USB CDC-ACM (USB serial). I believe it is currently configured for USB HID-only.

sago35, 2021-09-12:
The UART now works with PR #2107 .

> Also, let me know if you need any instructions to reconfigure that feature/usb-common branch to support USB CDC-ACM (USB serial). I believe it is currently configured for USB HID-only.

I'd like to know.

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram