tinygo-org/tinygo #2093: WebAssembly memory and --global-base

aduros, 2021-09-07:
I was wondering if there was a way to make TinyGo avoid using a reserved lower region of WebAssembly memory.

My initial attempt was to create a new target.json with these `wasm-ld` linker flags and use it to compile:

```json
{
    "inherits": [ "wasm" ],
    "ldflags": [
        "--import-memory",
        "--initial-memory=65536",
        "--max-memory=65536",
        "-zstack-size=7584",
        "--global-base=6560",
        "--strip-all"
    ]
}
```

This partially works, static data and the stack is correctly offset to the base region. But doing any kind of heap allocation seems to write data below the base region (somewhere in the first 1 KB of wasm memory).

For some context, this is for [WASM-4](https://github.com/aduros/wasm4), a virtual game console that reserves the first ~6 KB of memory for system data. If there was a way to do this, it would enable building games in Go.

See also:

- https://github.com/aduros/wasm4/issues/33
- https://github.com/aduros/wasm4/issues/66

Thank you!

aduros, 2021-09-14:
It looks like the wasm target uses the `--stack-first` linker flag, which seems to not work with `--global-base`.

I was able to workaround it by using this target.json to completely redefine the target instead of inheriting from the wasm target, in order to remove `--stack-first`:

```json
{
  "llvm-target": "wasm32--wasi",
  "build-tags": [ "tinygo.wasm" ],
  "goos": "js",
  "goarch": "wasm",
  "linker": "wasm-ld",
  "libc": "wasi-libc",
  "cflags": [
    "--target=wasm32--wasi",
    "--sysroot={root}/lib/wasi-libc/sysroot",
    "-Oz"
  ],
  "ldflags": [
    "--allow-undefined",
    "--no-demangle",
    "--import-memory",
    "--initial-memory=65536",
    "--max-memory=65536",
    "--global-base=6560",
    "-zstack-size=2048",
    "--strip-all"
  ],
  "wasm-abi": "js"
}
```

Would there be any side effects from not using `--stack-first`?

aykevl, 2021-09-15:
> Would there be any side effects from not using `--stack-first`?

I don't think so. The main benefit is that the WASM runtime will trap when the stack overflows (because it wraps around from 0 to the high end of memory) but that benefit is lost anyway when the lowest area is reserved.

However, wouldn't the default behavior work here? By default (with `--stack-first`), the stack is put in the first 64kB of memory. Because the stack goes from high to low addresses, the lowest area is normally untouched. You'd still have around 58kB of stack before it runs in the reserved area.