tinygo-org/tinygo #2099: Porting Guide

miathedev, 2021-09-08:
Is there any porting guide?

I would like to port tinygo to Heltecs Cubecell SoC. A doc explaining where i have to place which file would be really helpful.

With kind regards,
Mia

dgryski, 2021-09-08:
The wiki has some instructions on adding a new board: https://github.com/tinygo-org/tinygo/wiki/Adding-a-new-board

miathedev, 2021-09-08:
The Helltec Cubecell is based on ARM-Cortex-M4
https://github.com/HelTecAutomation/CubeCell-Arduino/tree/master/cores/asr6601/platform/CMSIS

When i take a look at the wiki - im pretty sure i wont succeed :/ I hopped its build upon arduino and not so low level.
Is someone able to please help me with this?