tinygo-org/tinygo #2106: Quick install guide does not work in Fedora 34

jens-r, 2021-09-10:
I have used "Quick install guide" (https://tinygo.org/getting-started/install/linux/#fedora-linux) and "Short introduction to TinyGo" (https://tinygo.org/getting-started/overview/) on Fedora 34, but it does not work. I've recreated the issue in Docker (Podman):

    podman run -it fedora:34
    dnf -y install tinygo avr-gcc avr-libc avrdude
    cat << EOF > hello.go
    package main
    
    import "fmt"
    
    func main() {
        fmt.Println("Hello world!")
    }
    EOF
    tinygo build -o hello ./hello.go

> error: failed to build src/runtime/gc_amd64.S: none of these commands were found in your $PATH: clang-11

I simplify the code and add "-target arduino" it works (however, it does not work on a Raspberry Pi with Fedora 34 (Fedora-Server-34-1.2.aarch64.raw.xz)).

```
package main
 
func main() {
    print("Hello world!")
}
```
    tinygo build -o hello -target arduino ./hello.go

The blinky1.go example does not work in Fedora 34 on a Raspberry Pi 4 (Fedora-Server-34-1.2.aarch64.raw.xz).

    wget https://raw.githubusercontent.com/tinygo-org/tinygo/release/src/examples/blinky1/blinky1.go
    tinygo build -o blinky1 -target arduino ./blinky1.go

> blinky1.go:6:2: package machine is not in GOROOT (/root/.cache/tinygo/goroot-go1.16.6-5d37adfc51f992721bdea3f3c8a97ffed0c7a13177334df3e15ce627842238c3-syscall/src/machine)

aykevl, 2021-09-15:
I suspect this is a bug in how the `tinygo` package is packaged in Fedora, not a bug in TinyGo itself. The `clang-11` command is in many cases needed for `tinygo` to work correctly. So I think this bug should be submitted to Fedora, not here.

QuLogic, 2021-09-27:
Fedora 34 has LLVM 12, which tinygo does not support, so I have been unable to update the package. It tries to use clang 11, but Fedora only has the libraries, and not the `clang` binary, so doesn't work. I haven't figured out any way to work around that.

QuLogic, 2022-02-08:
The [update](https://bodhi.fedoraproject.org/updates/FEDORA-2022-898397147b) is now out and heading to testing soon.

KopfKrieg, 2022-02-13:
I have encountered the same problem with Fedora 35, resulting in the exactly same error message. Any way to fix that? Fedora 35 ships with a lot of clang11-* packages, but it seems that the required binary itself is missing.

QuLogic, 2022-02-13:
Use the [Fedora 35 update](https://bodhi.fedoraproject.org/updates/FEDORA-2022-a1abd860fd).

KopfKrieg, 2022-02-14:
> Use the [Fedora 35 update](https://bodhi.fedoraproject.org/updates/FEDORA-2022-a1abd860fd).

Yep, that fixes it, thank you very much.

One note (not sure where to put this and if this is the right place to put it): the permissions for /usr/lib64/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0plus/picolibc/lib.a and /usr/lib64/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0plus/compiler-rt/lib.a seem to be wrong (both have 0600, but probably should have 0644, at least it works with that :sweat_smile: ).

QuLogic, 2022-02-19:
Ah, you should leave a comment on the update to prevent it going stable. Since it is stable, I'm going to close this issue and push a new update that fixes the permissions.

closed 2022-02-19 by QuLogic