tinygo-org/tinygo #2108: How to set max memory in wasm?

zhenjunMa, 2021-09-13:
### 1. background

In my case, i want to limit the max memory that the `*.wasm` file can use, if it reached the limit, it can crash by OOM.

### 2. what i have tried

i write a `a.go`, the content as follows:

```
package main

import (
	"fmt"
)

func main() {
	fmt.Println("hello")
}
```
then, i build this file by the command `tinygo build -o a.wasm -target=wasi a.go`

at last, i run `a.wasm` as follows:
```
> wasmtime a.wasm
hello
```

now, i decompiled `a.wasm` to `wat file` by command `wasm2wat a.wasm -o a`.

i found the memory segment in the wat file as folows:
[image]

`(memory (;0;) 2)` means the init memory is 2 pages, size is 65536 * 2 bytes, and no max memory limit, which means it can grow memory as needed.

i directly edit the wat file and modified the memory segment to `(memory (;0;) 1 1)`, which means the init memory 1 page and the max is 2 page,

now i compiled the wat file to wasm file by the command `wat2wasm a -o a.wasm` and run it by `wasmtime`

i got the error as follows
```
Error: failed to run main module `a.wasm`
Caused by:
    0: failed to invoke command default
    1: wasm trap: out of bounds memory access
```
it indeed what i want to see.

### 3. my question

in my opinion, maybe there is a method to set max memory limit before i build it.

i tried some method on the internet as follows
```
tinygo build -o function1.wasm -target=wasi -ldflags="-z stack-size=2048 --max-memory=65536" aa.go
```
but i got a error:
```
flag provided but not defined: -z
Usage of link:
  -X value
    	Set the value of the string variable to the given value.
```

when i remove the `-z` param like this:
```
tinygo build -o function1.wasm -target=wasi -ldflags="stack-size=2048 --max-memory=65536" aa.go
```

it succeed, but the memory segment is also `(memory (;0;) 2)`, nothing happened.

any idea about this? looking forward for your reply.

### 4. tinyGo version

`> tinygo version`
`tinygo version 0.18.0 darwin/amd64 (using go version go1.16 and LLVM version 11.0.0)`

### 5. more info

In my case,  i also want to limit the max cpu cycle used by wasm file,  maybe this is more difficult than limit max memory, any idea about this?

aduros, 2021-09-14:
I also had trouble using the `-ldflags` option. What seemed to work was compiling with `tinygo build ... -target ./target.json`, and putting something like this in target.json:

```json
{
    "inherits": [ "wasm" ],
    "ldflags": [
        "--initial-memory=65536",
        "--max-memory=65536",
        "-zstack-size=2048"
    ]
}
```

zhenjunMa, 2021-09-14:
@aduros 

cool!  it really works!

[image]

i found the memory segment is changed, now i just need to learn how to set cpu limit.

aykevl, 2021-09-15:
Yeah, there are not many people who want to limit memory in this way so I'm not sure whether it's worth it to add a flag for this feature. The `ldflags` as indicated above seems like a good way to do this for now.

> In my case, i also want to limit the max cpu cycle used by wasm file, maybe this is more difficult than limit max memory, any idea about this?

What do you mean? There is no way to do this in WebAssembly directly. This is something the embedder will have to do by counting instructions for example.

zhenjunMa, 2021-09-16:
> What do you mean? There is no way to do this in WebAssembly directly. This is something the embedder will have to do by counting instructions for example.

Right! i found some wasm runtime, e.g. wasmtime, provide API to limit the number of executable instructions in wasm.

deadprogram, 2024-11-27:
Closing due to question appears to have been answered. Thank you everyone!

closed 2024-11-27 by deadprogram