tinygo-org/tinygo #2109: SEEK_CUR not declared by package os

typeless, 2021-09-14:
I am trying to use tinygo to build a binary for an embedded Linux device.
I got the following errors:
```
$ tinygo build -o out.tiny .
../../cap/v4/cap.go:767:27: SEEK_CUR not declared by package os
../../cap/v4/cap.go:851:46: SEEK_SET not declared by package os
```
TinyGo version:
```
$ tinygo version
tinygo version 0.20.0-dev-ca4f251 linux/amd64 (using go version go1.17 and LLVM version 11.0.0)
```
My build OS:
```
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.3 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.3 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

ardnew, 2021-09-14:
It appears a package you are using depends on `os` from the Go standard library, which obviously isn't available for baremetal TinyGo.

Can you share the output of command `tinygo list -json` from the same directory you are trying to build from?

ardnew, 2021-09-14:
Also, if you are trying to build native executables for Ubuntu, see: https://tinygo.org/docs/guides/linux/ 

Otherwise, you need to specify a build target for tinygo, such as `tinygo build -o=out.tiny -target=arduino .`

typeless, 2021-09-15:
I tried with `GOARCH=arm` as mentioned in the guides, but it gave the same build error too.

This is a proprietary project, which contains some information that I am afraid unsuitable to share.

```
$ tinygo list -json
{
	"Dir": "/home/mural/src/castlestech.com/lbox/cmd/uld",
	"ImportPath": "castlestech.com/lbox/cmd/uld",
	"Name": "main",
	"Target": "/home/mural/bin/uld",
	"Root": "/home/mural/src/castlestech.com/lbox",
	"Module": {
		"Path": "castlestech.com/lbox",
		"Main": true,
		"Dir": "/home/mural/src/castlestech.com/lbox",
		"GoMod": "/home/mural/src/castlestech.com/lbox/go.mod",
		"GoVersion": "1.13"
	},
	"Match": [
		"."
	],
	"Stale": true,
	"StaleReason": "stale dependency: reflect",
	"GoFiles": [
		"main.go"
	],
	"Imports": [
		"castlestech.com/lbox/cmd/uld/internal/captype",
		"castlestech.com/lbox/cmd/uld/internal/decap",
		"castlestech.com/lbox/cmd/uld/internal/download",
		"castlestech.com/lbox/cmd/uld/internal/upload",
		"flag",
		"fmt",
		"io",
		"log",
		"os"
	],
	"Deps": [
		"bufio",
		"bytes",
		"castlestech.com/lbox/cap/v3",
		"castlestech.com/lbox/cap/v4",
		"castlestech.com/lbox/cmd/uld/internal/captype",
		"castlestech.com/lbox/cmd/uld/internal/decap",
		"castlestech.com/lbox/cmd/uld/internal/download",
		"castlestech.com/lbox/cmd/uld/internal/upload",
		"castlestech.com/lbox/dle",
		"castlestech.com/lbox/multitty",
		"castlestech.com/lbox/crypto/cipher",
		"castlestech.com/lbox/crypto/rsa",
		"context",
		"crypto",
		"crypto/aes",
		"crypto/cipher",
		"crypto/des",
		"crypto/dsa",
		"crypto/ecdsa",
		"crypto/ed25519",
		"crypto/ed25519/internal/edwards25519",
		"crypto/ed25519/internal/edwards25519/field",
		"crypto/elliptic",
		"crypto/elliptic/internal/fiat",
		"crypto/internal/randutil",
		"crypto/internal/subtle",
		"crypto/md5",
		"crypto/rand",
		"crypto/rsa",
		"crypto/sha1",
		"crypto/sha256",
		"crypto/sha512",
		"crypto/subtle",
		"crypto/x509",
		"crypto/x509/pkix",
		"device",
		"encoding/asn1",
		"encoding/base64",
		"encoding/binary",
		"encoding/hex",
		"encoding/pem",
		"errors",
		"flag",
		"fmt",
		"github.com/pkg/term",
		"github.com/pkg/term/termios",
		"hash",
		"internal/bytealg",
		"internal/cpu",
		"internal/fmtsort",
		"internal/itoa",
		"internal/oserror",
		"internal/race",
		"internal/reflectlite",
		"internal/task",
		"internal/unsafeheader",
		"io",
		"io/fs",
		"io/ioutil",
		"log",
		"math",
		"math/big",
		"math/bits",
		"math/rand",
		"net",
		"net/url",
		"os",
		"os/signal",
		"path",
		"path/filepath",
		"reflect",
		"runtime",
		"runtime/interrupt",
		"sort",
		"strconv",
		"strings",
		"sync",
		"sync/atomic",
		"syscall",
		"time",
		"unicode",
		"unicode/utf16",
		"unicode/utf8",
		"unsafe",
		"vendor/golang.org/x/crypto/cryptobyte",
		"vendor/golang.org/x/crypto/cryptobyte/asn1"
	],
	"TestGoFiles": [
		"download_test.go"
	],
	"TestImports": [
		"bufio",
		"context",
		"fmt",
		"io",
		"os",
		"os/exec",
		"regexp",
		"testing",
		"time"
	]
}
```

aykevl, 2021-09-15:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/2114

typeless, 2021-09-16:
#2114 Fixes the problem.

There are other missing APIs.
```
internal/decap/decap.go:201:16: Rename not declared by package os
internal/decap/decap.go:231:16: Chdir not declared by package os
internal/decap/decap.go:265:12: NewFile not declared by package os
```
Should I create a new issue?

aykevl, 2021-09-16:
You can do that, or create a PR to add these to the `os` package :)

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram