tinygo-org/tinygo #2124: Slow interpreting complex init functions

dkegel-fastly, 2021-09-19:
After applying #2001 to work around #1848, I appear to be running into something like #2090, but for trees.

Reproducer:
```
package main

import ac "github.com/cloudflare/ahocorasick"

var foo = ac.NewStringMatcher([]string{
        "axaxxxaxxaaa",
        "axaxxxaxxax",
        "xaxaxxxxaxxxx",
        "xaxaxxxxxxxaaqq",
        "xaxaxxxxxxxxxqq",
        "xaxxxxxxxaaqq",
        "xxaxaxaaax",
        "xxaxaxaaxaxa",
        "xxxaxaaaaaax",
        "xxxaxaaaaaaxqq",
        "xxxaxaaxqqqq",
        "xxxxaaaxaxa",
        "xxxxaaxxxx",
})

func main() {
	println("test")
}
```

```
$ time go build
real	0m0.215s
user	0m0.163s

$ time tinygo build -o foo.wasm -target wasm main.go
real	0m11.913s
user	0m32.456s
```

Adding more strings to the list causes compile time to increase rapidly.

deadprogram, 2021-11-10:
How is this issue with the latest `dev` branch? Same problem still exist?

dgryski, 2021-11-10:
Yes, interp still causes timeouts creating large data structures at build time.

aykevl, 2021-11-11:
This does sound very much like a duplicate of #2090. Maybe the cause is different, but it certainly sounds like a duplicate.

dkegel-fastly, 2021-11-11:
Yes.  Interesting it can be triggered with so few entries.

dgryski, 2021-11-11:
There are so few entries in this case because `interp` is trying to run `ahocorasick.NewStringMatcher()` and that's taking the time.  Same reason global regexps timeout.

dankegel, 2022-05-01:
Dup of 2090, whose title might want to change slightly.

dkegel-fastly, 2022-05-04:
Closing as dup of #2090.

closed 2022-05-04 by dkegel-fastly