tinygo-org/tinygo #2131: Question: Do TinyGo heap allocations and C malloc() map to the same thing?

bradleypeabody, 2021-09-22:
A quick question that I couldn't see easily from the docs or source:  In trying to optimize memory for a program that is a mix of Go and C code (using LittleFS based on https://github.com/tinygo-org/tinyfs) the question came up of is there any difference between C `malloc()` calls vs performing a heap allocation in Go?  I assume there would be since `malloc()` is not GC'ed whereas Go `make()` calls that end up using the heap are, but I wasn't sure from a memory efficiency perspective (size specifically I'm referring to, not speed) if there is an advantage to removing all C `malloc()` calls from the program so the heap is entirely controlled by the Go runtime vs having to share it with whatever C `malloc()` is doing.

aykevl, 2021-09-24:
(I was answering this but the comment sadly got lost)

The short answer is: it depends. On baremetal and WebAssembly, the TinyGo heap is also used for the C heap. But for hosted platforms (Linux, MacOS), the libc of that host is used for `malloc` and the TinyGo heap is separate.

I would suggest using the heap that is more appropriate for what you're doing. If the memory is mainly used from C, use `malloc`. If the memory is stored in Go but only shared temporarily with C during a call, use regular Go memory.

Another consideration is that Go heap allocations can sometimes be optimized (to stack allocations, for example). C malloc is not currently optimized in any way.

aykevl, 2021-09-24:
>  I assume there would be since `malloc()` is not GC'ed whereas Go `make()` calls that end up using the heap are

In TinyGo, `malloc` is also allocated on the TinyGo heap and so will also be GC'd. The `free` function is currently a no-op (but may be implemented in the future).

bradleypeabody, 2021-09-24:
Thanks, yeah that explains it well.  Closing the issue.

Just for posterity, as of this writing the current implementation called by C malloc()s is in src/runtime/baremetal.go:

```
//export malloc
func libc_malloc(size uintptr) unsafe.Pointer {
        return alloc(size)
}
```

closed 2021-09-24 by bradleypeabody