tinygo-org/tinygo #2132: brew install tiny failure macOS

davidgs, 2021-09-22:
```
$ brew tap --verbose --debug tinygo-org/tools
/usr/local/Homebrew/Library/Homebrew/shims/scm/git --version
==> Tapping tinygo-org/tools
git clone https://github.com/tinygo-org/homebrew-tools /usr/local/Homebrew/Library/Taps/tinygo-org/homebrew-tools --origin=origin --template=
Cloning into '/usr/local/Homebrew/Library/Taps/tinygo-org/homebrew-tools'...
fatal: invalid branch name: init.defaultBranch =
Error: Failure while executing; `git clone https://github.com/tinygo-org/homebrew-tools /usr/local/Homebrew/Library/Taps/tinygo-org/homebrew-tools --origin=origin --template=` exited with 128.
/usr/local/Homebrew/Library/Homebrew/utils.rb:322:in `safe_system'
/usr/local/Homebrew/Library/Homebrew/tap.rb:299:in `install'
/usr/local/Homebrew/Library/Homebrew/cmd/tap.rb:65:in `tap'
/usr/local/Homebrew/Library/Homebrew/brew.rb:129:in `<main>'
$
```
Which, then, of course `brew install tinygo` fails too.

deadprogram, 2021-09-22:
Perhaps you need to run `git config --global init.defaultBranch master`? Looks like your version of git that not use any default branch. Looks like https://github.com/Homebrew/brew/issues/10766

davidgs, 2021-09-23:
That was it, though I chose `main` as the default branch. Still worked. Thanks!

closed 2021-09-23 by davidgs