tinygo-org/tinygo #2141: sort.Slice cause memory corruption on custom data struct

learnforpractice, 2021-09-26:
```go
package main

import (
	"encoding/hex"
	"fmt"
	"sort"
)

func less(a, b []byte) bool {
	for i := 0; i < len(a) && i < len(b); i++ {
		if a[i] < b[i] {
			return true
		}
		if a[i] > b[i] {
			return false
		}
	}
	return len(a) < len(b)
}

type MyData struct {
	Type int
	Data [33]byte
}

func main() {
	aa := make([]MyData, 3)
	data, _ := hex.DecodeString("02cbb8488ce044116dd4411c7c76ca44bb42b76400a0eb8353bf62a327423da26c")
	aa[0].Type = 0
	copy(aa[0].Data[:], data)

	data, _ = hex.DecodeString("02056f8d91fc8996bd3f2fa4dc2dd60d6cdd05811db0065dd4f191b8bbd8eda5d4")
	aa[1].Type = 0
	copy(aa[1].Data[:], data)

	data, _ = hex.DecodeString("039cd3ad335ada89a15806f2b06e36f938f0b2b798ca4378b985135297211b9de3")
	aa[2].Type = 0
	copy(aa[2].Data[:], data)

	sort.Slice(aa, func(i, j int) bool {
		// return bytes.Compare(aa[i].Data[:], aa[j].Data[:]) < 0
		return less(aa[i].Data[:], aa[j].Data[:])
	})

	for _, a := range aa {
		fmt.Printf("%d %x\n", a.Type, a.Data[:])
	}
}
```

expected output:
```
0 02056f8d91fc8996bd3f2fa4dc2dd60d6cdd05811db0065dd4f191b8bbd8eda5d4
0 02cbb8488ce044116dd4411c7c76ca44bb42b76400a0eb8353bf62a327423da26c
0 039cd3ad335ada89a15806f2b06e36f938f0b2b798ca4378b985135297211b9de3
```

got:
```
0 0000000000000002056f8d91fc8996bd3f2fa4dc2dd60d6cdd05811db0065dd4f1
0 00000000000000039cd3ad335ada89a15806f2b06e36f938f0b200000000000000
8537730062649987396 ca44bb42b76400a0eb8353bf62a327423da26cb798ca4378b985135297211b9de3
```

dgryski, 2021-09-26:
I've managed to shrink this down a bit; it looks like the memcpy call in `reflect.Swapper` is getting the wrong size.  This is because it doesn't take into account the padding for individual array elements.  For example,

```
package main

import (
	"reflect"
	"unsafe"
)

type MyData struct {
	Type int
	Data [1]byte
}

func main() {
	aa := []MyData{
		{1, [1]byte{0x01}},
		{2, [1]byte{0x02}},
	}
	println(uintptr(unsafe.Pointer(&aa[0])))
	println(uintptr(unsafe.Pointer(&aa[1])))

	v := reflect.ValueOf(aa)
	println(v.Index(0).Type().Size())
}
```

produces the output
```
0x000000010f5b7000
0x000000010f5b7010
0x0000000000000009
```

However, instrumenting `reflect.Swapper` to print out the memory locations to be swapped
```
diff --git a/src/reflect/swapper.go b/src/reflect/swapper.go
index 82842de9..29b56914 100644
--- a/src/reflect/swapper.go
+++ b/src/reflect/swapper.go
@@ -33,6 +33,7 @@ func Swapper(slice interface{}) func(i, j int) {
                }
                val1 := unsafe.Pointer(uintptr(header.data) + uintptr(i)*size)
                val2 := unsafe.Pointer(uintptr(header.data) + uintptr(j)*size)
+               println(val1, val2)
                memcpy(tmp, val1, size)
                memcpy(val1, val2, size)
                memcpy(val2, tmp, size)
```

and rerunning we see that the actual locations copied are missing the intra-element padding.
```
0x0000000111ac4000
0x0000000111ac4010
0x0000000000000009
0x0000000111ac4000 0x0000000111ac4009
```

So, the actual address of `aa[1]` is `0x0000000111ac4010` but instead we copy from `0x0000000111ac4009`.

The size calculation should obviously take into consideration the alignment of underlying type.

I'm not sure if 
```
        size := align(typ.Size(), uintptr(typ.Align()))
```
is the correct fix here, or if there's another/better way to get this information.

learnforpractice, 2021-09-26:
I think typ.Size() should return the aligned size directly.

learnforpractice, 2021-09-26:
change code at 
https://github.com/tinygo-org/tinygo/blob/157382600535c95bdbf5d6e8c4b9e5ca71749018/src/reflect/type.go#L551
to
```go
	case Struct:
		numField := t.NumField()
		if numField == 0 {
			return 0
		}
		lastField := t.rawField(numField - 1)
		return align(lastField.Offset+lastField.Type.Size(), uintptr(t.Align()))
```
fix the problem.

dgryski, 2021-09-26:
Getting the size wrong here via `reflect` also affects `reflect.Value.Index()`, and probably other thing that are trying to calculate offsets manually.  We need to ensure the fix that goes into `reflect.Type` matches how the compiler adds padding.

aykevl, 2021-09-26:
@learnforpractice yes this looks like a correct fix. It matches what the compiler does:

https://github.com/tinygo-org/tinygo/blob/157382600535c95bdbf5d6e8c4b9e5ca71749018/compiler/sizes.go#L147-L148

dgryski, 2021-09-26:
@learnforpractice I can file a PR with this fix if you'd like.

learnforpractice, 2021-09-27:
Sure, it’s good to know your confirmation

PR #2142 (2021-09-27): src/reflect: fix type.Size() to account for struct padding
Fixes #2141
commit d9ad500c (2021-09-27): src/reflect: fix type.Size() to account for struct padding

Fixes #2141
closed 2021-11-18 by aykevl