tinygo-org/tinygo #2158: [Question] Does TinyGo Intend on Supporting Generics?

danforbes, 2021-10-04:
[Generics are coming to Go](https://github.com/golang/go/issues/43651) 🚀 Does TinyGo plan on supporting this long-awaited feature?

soypat, 2021-10-09:
From what I've heard on the #tinygo Slack channel from experienced TinyGo developers, generics are a welcome feature. I had heard @ardnew mention generics would make writing USB CDC drivers much easier, if I recall correctly.

ibudisteanu, 2022-01-25:
This would be great! For example, the package I am helping with already uses generics. Without support for generics would introduce a barrier to use TinyGo in order to reduce WASM bundles sizes. We will wait for Generics support in tinygo.

deadprogram, 2022-06-11:
See https://github.com/tinygo-org/tinygo/pull/2903

closed 2022-07-01 by deadprogram