tinygo-org/tinygo #2172: infinite recursion trying to create recursive function type

dgryski, 2021-10-13:
Tinygo does not like `fnType` in the below program:

```
package main

type Node struct {
	v    int
	next *Node
}

type fnType func(*Node) (fnType, *Node, error)

func printNode(n *Node) (fnType, *Node, error) {
	println(n.v)
	return nil, n.next, nil
}

func main() {

	node := Node{
		v:    0,
		next: nil,
	}

	for fn, n, err := printNode(&node); err == nil && fn != nil; {
		fn, n, err = fn(n)
	}
}
```

tinygo errors with a stack overflow; the callstack is filled with
```
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/compiler/func.go:99 +0x8c fp=0xc020d40560 sp=0xc020d40520 pc=0x429c7ec
github.com/tinygo-org/tinygo/compiler.(*compilerContext).makeLLVMType(0xc0005b0000, {0x4413240, 0xc000ca67e0})
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:388 +0x2b4 fp=0xc020d40628 sp=0xc020d40560 pc=0x4284cd4
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getLLVMType(0xc0005b0000, {0x4413240, 0xc000ca67e0})
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:321 +0x6e fp=0xc020d40668 sp=0xc020d40628 pc=0x428498e
github.com/tinygo-org/tinygo/compiler.(*compilerContext).makeLLVMType(0xc0005b0000, {0x44131f0, 0xc000128e80})
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:383 +0x771 fp=0xc020d40730 sp=0xc020d40668 pc=0x4285191
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getLLVMType(0xc0005b0000, {0x44131f0, 0xc000128e80})
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:321 +0x6e fp=0xc020d40770 sp=0xc020d40730 pc=0x428498e
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getRawFuncType(0xc0005b0000, 0xc000ca67e0)
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/compiler/func.go:125 +0x4cf fp=0xc020d40828 sp=0xc020d40770 pc=0x429cd4f
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getFuncType(0xc0005b0000, 0x0)
```

dgryski, 2021-10-13:
Interesting; this doesn't fail on `dev`, only on my local branch consisting of `dev` + #2011 + #2171

dgryski, 2021-10-13:
This triggers on `dev` when building for native.  The coroutine-logic prevents it from hitting wasm on dev, but *does* cause it to trigger for asyncify.

aykevl, 2021-10-13:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/2175

deadprogram, 2021-11-18:
I think the request was fulfilled in the new release, so closing. Please reopen if needed. Thanks!

closed 2021-11-18 by deadprogram