tinygo-org/tinygo #2185: Failed to compile github.com/hajimehoshi/ebiten/v2/examples/2048

hajimehoshi, 2021-10-16:
```
git clone https://github.com/hajimehoshi/ebiten
cd ebiten
docker pull tinygo/tinygo:latest
docker run --rm -v $(pwd):/src -w /src tinygo/tinygo:latest tinygo build -o wasm.wasm -target=wasm -tags=example ./examples/2048/
```

```
$ docker run --rm -v $(pwd):/src -w /src tinygo/tinygo:latest tinygo build -o wasm.wasm -target=wasm -tags=example ./examples/2048/                                                                                                                                                   
go: downloading golang.org/x/image v0.0.0-20210628002857-a66eb6448b8d                                                                                                                                                                                                                 
go: downloading golang.org/x/text v0.3.6                                                                                                                                                                                                                                              
runtime: goroutine stack exceeds 1000000000-byte limit                                                                                                                                                                                                                                
runtime: sp=0xc03e738398 stack=[0xc03e738000, 0xc05e738000]                                                                                                                                                                                                                           
fatal error: stack overflow                                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                                                                      
runtime stack:                                                                                                                                                                                                                                                                        
runtime.throw({0x4488e9e, 0x5f0a740})                                                                                                                                                                                                                                                 
        /usr/local/go/src/runtime/panic.go:1198 +0x71                                                                                                                                                                                                                                 
runtime.newstack()                                                                                                                                                                                                                                                                    
        /usr/local/go/src/runtime/stack.go:1088 +0x5ac                                                                                                                                                                                                                                
runtime.morestack()                                                                                                                                                                                                                                                                   
        /usr/local/go/src/runtime/asm_amd64.s:461 +0x8b                                                                                                                                                                                                                               
                                                                                                                                                                                                                                                                                      
goroutine 24 [running]:                                                                                                                                                                                                                                                               
runtime.cgoCheckUnknownPointer(0xc01da15400, {0x4497a51, 0x29})                                                                                                                                                                                                                       
        /usr/local/go/src/runtime/cgocall.go:554 +0x272 fp=0xc03e7383a8 sp=0xc03e7383a0 pc=0xa6c3d2                                                                                                                                                                                   
runtime.cgoCheckArg(0x441f6c0, 0xc01da15400, 0x60, 0x1, {0x4497a51, 0x29})                                                                                                                                                                                                            
        /usr/local/go/src/runtime/cgocall.go:546 +0x1c5 fp=0xc03e738428 sp=0xc03e7383a8 pc=0xa6bd65                                                                                                                                                                                   
runtime.cgoCheckPointer({0x441f6c0, 0xc01da15400}, {0x0, 0x0})                                                                                                                                                                                                                        
        /usr/local/go/src/runtime/cgocall.go:427 +0x156 fp=0xc03e738470 sp=0xc03e738428 pc=0xa6bb36                                                                                                                                                                                   
tinygo.org/x/go-llvm.ConstArray.func1({0xd8ad50}, 0xc03e7384e0, 0x3e7384e0)                                                                                                                                                                                                           
        /go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20210325115028-e7b85195e81c/ir.go:889 +0x53 fp=0xc03e7384b8 sp=0xc03e738470 pc=0xc5b1b3                                                                                                                                               
tinygo.org/x/go-llvm.ConstArray({0x7f2311aa4858}, {0xc01da15400, 0x0, 0x7f2319683440})                                                                                                                                                                                                
        /go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20210325115028-e7b85195e81c/ir.go:889 +0x2e fp=0xc03e7384e0 sp=0xc03e7384b8 pc=0xc5b10e                                                                                                                                               
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue.func1()                                                                                                                                                                                                                     
        /home/circleci/project/interp/memory.go:793 +0xbb fp=0xc03e738548 sp=0xc03e7384e0 pc=0xd28c1b                                                                                                                                                                                 
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue({{0xc014007c00, 0x2b1, 0xc03e7386c8}}, 0xc0036fb740)                                                                                                                                                                        
[... cut ...]
github.com/tinygo-org/tinygo/builder.jobWorker(0x0, 0x0)
        /home/circleci/project/builder/jobs.go:179 +0x5e
created by github.com/tinygo-org/tinygo/builder.runJobs
        /home/circleci/project/builder/jobs.go:101 +0x1a5

goroutine 23 [chan receive, 3 minutes]:
github.com/tinygo-org/tinygo/builder.jobWorker(0x0, 0x0)
        /home/circleci/project/builder/jobs.go:179 +0x5e
created by github.com/tinygo-org/tinygo/builder.runJobs
        /home/circleci/project/builder/jobs.go:101 +0x1a5
```

deadprogram, 2021-11-05:
Now that #2001 has been merged into the `dev` branch, any way to try this again @hajimehoshi? Thank you!

hajimehoshi, 2021-11-06:
Is this in the latest Docker image?

deadprogram, 2021-11-06:
Latest `dev` image, yes. `docker pull tinygo/tinygo-dev:latest` should work.

hajimehoshi, 2021-11-06:
Looks like the issue I reported fixed. Thanks!

closed 2021-11-06 by hajimehoshi