tinygo-org/tinygo #2186: stub syscall.Environ and syscall. SysProcAttr so stdlib tests using internal/testenv can build

dgryski, 2021-10-16:


aykevl, 2021-10-17:
I think implementing `syscall.Environ` wouldn't be very difficult (see `syscall.Getenv` for example), so I would prefer if it was fully implemented from the start.

dgryski, 2021-11-10:
Building our own `internal/testenv` might make more sense.   It's a pretty small package that we can easily tailor to testing needs.  However, since many of the tests look for particular features and then call `t.Skip()` (which isn't yet supported),  the runtime behaviour of the tests may not be what we expect if even they can now build.

dkegel-fastly, 2021-11-12:
This would help github.com/getlantern/golog build.  (Along with https://github.com/go-stack/stack/issues/31.)

Given that I ran into it really quickly in my little fishing expedition, this seems high value.

dgryski, 2021-12-08:
This was handled in https://github.com/tinygo-org/tinygo/pull/2268

@dkegel-fastly If you're still having trouble with getlantern/golog or go-stack/stack, let's open new bugs with what's actually missing.

closed 2021-12-08 by dgryski