tinygo-org/tinygo #2190: Dividing by zero should panic

learnforpractice, 2021-10-20:
Giving the following code
```go
package main

import "os"

func main() {
	a := 0
	if len(os.Args) == 1000 {
		a = 1 //never goes here， just to make sure the code can be compiled
	}
	println(1/a)
}
```

Compiling the code with go compiler and  run it, it will throw the following error

```
panic: runtime error: integer divide by zero
```

But after compiling it with tinygo, it just outputs "1".

aykevl, 2021-10-20:
You are correct, the spec says that a divide by zero must cause a runtime panic. Right now we use whatever LLVM does, which matches C (meaning, it is undefined behavior to divide by zero). This will need to be fixed.

A related issue is signed integer divide overflow, when you divide int32(-2147483648) by -1 this is undefined behavior in C and LLVM but defined for Go: it must result in the number -2147483648.

aykevl, 2021-10-20:
I've made a fix here: https://github.com/tinygo-org/tinygo/pull/2195
The code size impact of it is surprisingly small.

Thank you for reporting this issue!

deadprogram, 2021-11-18:
Released with 0.21 so now closing. Thank you!

closed 2021-11-18 by deadprogram