tinygo-org/tinygo #2232: Arduino Uno: will a simplest program fit the memory?

jakub-m, 2021-11-05:
[Arduino Uno][ref_uno] mentioned in [tinygo docs][ref_docs] has 32K memory. The simplest program I compile with tinygo gives 33K. Stupid question: will this even fit? Does the binary contain some unnecessary information (e.g. symbols) that can be stripped?

[ref_uno]:https://www.microchip.com/en-us/product/ATmega328p
[ref_docs]:https://tinygo.org/docs/reference/microcontrollers/arduino/

Example file:

```
package main

func main() {
    a := 1
    a = a + 2
}
```

build with `tinygo build -target=arduino -o main main.go` results in 33K binary.

ysoldak, 2021-11-06:
@jakub-m Oh yes, such little program definitely fits UNO.
You just have to build it bit differently, please specify format of the output file, like this:
`tinygo build -target=arduino -o main.bin main.go`
and resulting size then is just `764` bytes!

jakub-m, 2021-11-06:
Thanks @ysoldak for the awesome responsiveness!

closed 2021-11-06 by jakub-m