tinygo-org/tinygo #2234: nrf52840: floats do not work anymore

ysoldak, 2021-11-06:
Simple program as below would not work on nrf52840 chip anymore. Affected board is for ex. Nano 33 BLE.
```
package main
import "time"
func main() {
	x := 100.1
	time.Sleep(5 * time.Second)
	for {
		println(x)
		time.Sleep(time.Second)
	}
}
```
Commit that broke things is this:
https://github.com/tinygo-org/tinygo/commit/29206cf0a4a8efcdabe50306090fe6125f6b2d0c

If I remove `"cpu": "cortex-m4",` line from `cortex-m4.json` target file, it works again.

aykevl, 2021-11-06:
Thanks for the report! I think I know what's wrong and how to fix it. I'll take a look soon.

deadprogram, 2021-11-07:
Fixed by https://github.com/tinygo-org/tinygo/pull/2235 in cycle, so now closing.

closed 2021-11-07 by deadprogram