tinygo-org/tinygo #2236: Failed to compile github.com/hajimehoshi/ebiten/v2/examples/2048

hajimehoshi, 2021-11-06:
```sh
git clone https://github.com/hajimehoshi/ebiten
cd ebiten
docker pull tinygo/tinygo:latest
docker run --rm -v $(pwd):/src -w /src tinygo/tinygo-dev:latest tinygo build -o wasm.wasm -target=wasm -tags=example ./examples/2048/
```

```
go: downloading golang.org/x/image v0.0.0-20210628002857-a66eb6448b8d
go: downloading golang.org/x/text v0.3.6
panic: interp: load from object with external store

goroutine 9 [running]:
github.com/tinygo-org/tinygo/interp.(*memoryView).load(0xc00cd00270, {0xc000000cef}, 0x4)
        /tinygo/interp/memory.go:273 +0x2a5
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc017c80120, 0xc00a4a19f0, {0xc00d24db10, 0x4, 0x2}, 0xc01ae4e810, {0xc00d069428, 0x8})
        /tinygo/interp/interpreter.go:395 +0x4eae
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc017c80120, 0xc00a4a19a0, {0x0, 0x0, 0x0}, 0x0, {0x7e025d, 0x4})
        /tinygo/interp/interpreter.go:490 +0x8555
github.com/tinygo-org/tinygo/interp.Run({0xc018681dc8}, 0x5f)
        /tinygo/interp/interp.go:112 +0x630
github.com/tinygo-org/tinygo/builder.optimizeProgram({0x7f62ce14d550}, 0xc000070ea0)
        /tinygo/builder/build.go:761 +0x33
github.com/tinygo-org/tinygo/builder.Build.func2(0x0)
        /tinygo/builder/build.go:456 +0x66d
github.com/tinygo-org/tinygo/builder.jobWorker(0x0, 0x0)
        /tinygo/builder/jobs.go:191 +0xad
created by github.com/tinygo-org/tinygo/builder.runJobs
        /tinygo/builder/jobs.go:110 +0x229
```

Zyko0, 2021-11-09:
Same issue here

aykevl, 2021-11-11:
I have a fix for this issue (and another issue that came up after that), will make PRs soon.

aykevl, 2021-11-11:
Submitted the two PRs. With these two PRs merged a regular build should be able to build the 2048 example, but a build with LLVM assertions enabled will still panic because of https://bugs.llvm.org/show_bug.cgi?id=49916 (fixed in LLVM but not in LLVM 11).

deadprogram, 2021-11-11:
Both PRs have now been merged into the `dev` branch. As soon as it is done building, can someone please retest this?

Zyko0, 2021-11-11:
Just tested, and it built ! Good job ! 🚀

deadprogram, 2021-11-18:
I think the request was fulfilled in the new release, so closing. Please reopen if needed. Thanks!