tinygo-org/tinygo #2237: Unneccesary alloc in Hot path

oflebbe, 2021-11-06:
I noticed that there are IMO unnecessary allocations in the hot path if one uses the st7789 SPI driver on rp pico .
Looks like it might slow down SPI transmission considerably.

In order to debug it, I stripped down the code to the bare minimum still showing the diagnostics.
Please use tinygo (from dev branch) and compile
`tinygo build -o a    -print-allocs=. main.go
...main.go:14:14: object allocated on the heap: escapes at line 14
`
If you comment the switch or the print statement this warning goes away.
[main.txt](https://github.com/tinygo-org/tinygo/files/7490833/main.txt)

oflebbe, 2021-11-06:
For easier read here the source code inline
```go
package main

type MockSPI int

func (m MockSPI) Tx(w, r []byte) (err error) {
	switch {
	case w == nil:
		print(1)
	}
	return err
}

func (d MockSPI) SetWindow() {
	d.Tx([]uint8{1}, nil)
}

func main() {
	var m MockSPI = 1
	m.SetWindow()
}
```

dgryski, 2024-10-21:
So it's just the slice header here.  If we replace the slice with a pointer to an array then we're ok.
```
~/go/src/github.com/dgryski/bug/spialloc $ cat main.go
package main

type MockSPI int

func (m MockSPI) Tx(w *[1]byte, r []byte) (err error) {
	switch {
	case w == nil:
		print(1)
	}
	return err
}

func (d MockSPI) SetWindow() {
	d.Tx(&[...]byte{1}, nil)
}

func main() {
	var m MockSPI = 1
	m.SetWindow()
}

```

This seems like the sort of thing we ought to be able to fix.

dgryski, 2024-10-21:
This is what the Go SSA looks like for `SetWindow`
```
func (d MockSPI) SetWindow():
0:                                                                entry P:0 S:0
	t0 = new [1]byte (slicelit)                                    *[1]byte
	t1 = &t0[0:int]                                                   *byte
	*t1 = 1:byte
	t2 = slice t0[:]                                                 []byte
	t3 = (MockSPI).Tx(d, t2, nil:[]byte)                              error
	return
```

aykevl, 2024-10-22:
Also, I've fixed this specific issue a while ago in the driver: https://github.com/tinygo-org/drivers/pull/545

deadprogram, 2024-12-04:
Closing since was addressed. Thanks everyone!

closed 2024-12-04 by deadprogram