tinygo-org/tinygo #2238: ld.lld-11 undefined symbol time.stopTimer / time.startTimer

cuongvo, 2021-11-06:
Getting an odd error on both the current dev branch via docker image:

```
root@e0409d9a0aba:/code/experiment/wasm# tinygo build -o rk main.go
ld.lld-11: error: undefined symbol: time.stopTimer
>>> referenced by sleep.go:81 (/usr/local/go/src/time/sleep.go:81)
>>>               /tmp/tinygo800274430/main.o:((*context.timerCtx).cancel)

ld.lld-11: error: undefined symbol: time.startTimer
>>> referenced by sleep.go:175 (/usr/local/go/src/time/sleep.go:175)
>>>               /tmp/tinygo800274430/main.o:(context.WithTimeout)
error: failed to link /tmp/tinygo800274430/main: exit status 1
```

and release 0.20.0 via docker image:

```
root@0facfb37e223:/code/experiment/wasm# tinygo build -o rk main.go
/usr/bin/ld: /tmp/tinygo3807953377/main.o: in function `(*time.Timer).Stop':
/usr/local/go/src/time/sleep.go:81: undefined reference to `time.stopTimer'
/usr/bin/ld: /tmp/tinygo3807953377/main.o: in function `time.AfterFunc':
/usr/local/go/src/time/sleep.go:175: undefined reference to `time.startTimer'
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo3807953377/main: exit status 1
```

And the same error on an M1 Mac with 0.20.0:

```
Undefined symbols for architecture x86_64:
  "_time.startTimer", referenced from:
      _context.WithTimeout in main.o
  "_time.stopTimer", referenced from:
      _(*context.timerCtx).cancel in main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
error: failed to link /var/folders/c0/2qkwprys5t797t_k7dx_djs00000gn/T/tinygo1349519740/main: exit status 1
```
This same program builds as a wasm target with no error, but includes `time.stopTimer` and `time.startTimer` as import targets:

```
Type: wasm
Size: 2.3 MB
Imports:
  Functions:
    "wasi_snapshot_preview1"."fd_write": [I32, I32, I32, I32] -> [I32]
    "env"."runtime.ticks": [] -> [F64]
    "env"."syscall/js.valuePrepareString": [I32, I32, I32, I32] -> []
    "env"."syscall/js.valueLoadString": [I32, I32, I32, I32, I32, I32] -> []
    "env"."syscall/js.finalizeRef": [I32, I32, I32] -> []
    "env"."syscall/js.stringVal": [I32, I32, I32, I32, I32] -> []
    "env"."time.startTimer": [I32, I32, I32] -> []
    "env"."syscall/js.valueIndex": [I32, I32, I32, I32, I32] -> []
    "env"."syscall/js.valueCall": [I32, I32, I32, I32, I32, I32, I32, I32, I32] -> []
    "env"."syscall/js.valueLength": [I32, I32, I32] -> [I32]
    "env"."syscall/js.valueSetIndex": [I32, I32, I32, I32, I32] -> []
    "env"."syscall/js.valueNew": [I32, I32, I32, I32, I32, I32, I32] -> []
    "env"."syscall/js.valueSet": [I32, I32, I32, I32, I32, I32] -> []
    "env"."time.stopTimer": [I32, I32, I32] -> [I32]
    "env"."syscall/js.valueGet": [I32, I32, I32, I32, I32, I32] -> []
    "wasi_snapshot_preview1"."proc_exit": [I32] -> []
  Memories:
  Tables:
  Globals:
Exports:
  Functions:
    "malloc": [I32] -> [I32]
    "free": [I32] -> []
    "calloc": [I32, I32] -> [I32]
    "realloc": [I32, I32] -> [I32]
    "posix_memalign": [I32, I32, I32] -> [I32]
    "aligned_alloc": [I32, I32] -> [I32]
    "malloc_usable_size": [I32] -> [I32]
    "_start": [] -> []
    "resume": [] -> []
    "go_scheduler": [] -> []
  Memories:
    "memory": not shared (3 pages..)
  Tables:
  Globals:
```

cuongvo, 2021-11-07:
Digging into this more:

```
Type: wasm
Size: 1.4 MB
Imports:
  Functions:
    "wasi_snapshot_preview1"."fd_write": [I32, I32, I32, I32] -> [I32]
    "wasi_snapshot_preview1"."clock_time_get": [I32, I64, I32] -> [I32]
    "wasi_snapshot_preview1"."args_sizes_get": [I32, I32] -> [I32]
    "wasi_snapshot_preview1"."args_get": [I32, I32] -> [I32]
    "env"."time.stopTimer": [I32, I32, I32] -> [I32]
    "env"."time.startTimer": [I32, I32, I32] -> []
    "wasi_snapshot_preview1"."fd_read": [I32, I32, I32, I32] -> [I32]
    "wasi_snapshot_preview1"."fd_write": [I32, I32, I32, I32] -> [I32]
    "wasi_snapshot_preview1"."fd_close": [I32] -> [I32]
    "wasi_snapshot_preview1"."fd_fdstat_get": [I32, I32] -> [I32]
    "wasi_snapshot_preview1"."path_open": [I32, I32, I32, I32, I32, I64, I64, I32, I32] -> [I32]
    "wasi_snapshot_preview1"."fd_prestat_get": [I32, I32] -> [I32]
    "wasi_snapshot_preview1"."fd_prestat_dir_name": [I32, I32, I32] -> [I32]
    "wasi_snapshot_preview1"."proc_exit": [I32] -> []
    "wasi_snapshot_preview1"."environ_sizes_get": [I32, I32] -> [I32]
    "wasi_snapshot_preview1"."environ_get": [I32, I32] -> [I32]
  Memories:
  Tables:
  Globals:
Exports:
  Functions:
    "malloc": [I32] -> [I32]
    "free": [I32] -> []
    "calloc": [I32, I32] -> [I32]
    "realloc": [I32, I32] -> [I32]
    "posix_memalign": [I32, I32, I32] -> [I32]
    "aligned_alloc": [I32, I32] -> [I32]
    "malloc_usable_size": [I32] -> [I32]
    "_start": [] -> []
    "math.Acosh": [F64, I32, I32] -> [F64]
    "math.Log": [F64, I32, I32] -> [F64]
    "math.Sqrt": [F64, I32, I32] -> [F64]
    "math.Log1p": [F64, I32, I32] -> [F64]
    "math.Frexp": [I32, F64, I32, I32] -> []
    "math.Asin": [F64, I32, I32] -> [F64]
    "math.Acos": [F64, I32, I32] -> [F64]
    "math.Asinh": [F64, I32, I32] -> [F64]
    "math.Atan": [F64, I32, I32] -> [F64]
    "math.Atan2": [F64, F64, I32, I32] -> [F64]
    "math.Atanh": [F64, I32, I32] -> [F64]
    "math.Cbrt": [F64, I32, I32] -> [F64]
    "math.Max": [F64, F64, I32, I32] -> [F64]
    "math.Min": [F64, F64, I32, I32] -> [F64]
    "math.Erf": [F64, I32, I32] -> [F64]
    "math.Exp": [F64, I32, I32] -> [F64]
    "math.Ldexp": [F64, I32, I32, I32] -> [F64]
    "math.Erfc": [F64, I32, I32] -> [F64]
    "math.Exp2": [F64, I32, I32] -> [F64]
    "math.Expm1": [F64, I32, I32] -> [F64]
    "math.Floor": [F64, I32, I32] -> [F64]
    "math.Modf": [I32, F64, I32, I32] -> []
    "math.Ceil": [F64, I32, I32] -> [F64]
    "math.Trunc": [F64, I32, I32] -> [F64]
    "math.Pow": [F64, F64, I32, I32] -> [F64]
    "math.Sin": [F64, I32, I32] -> [F64]
    "math.Hypot": [F64, F64, I32, I32] -> [F64]
    "math.Cos": [F64, I32, I32] -> [F64]
    "math.Mod": [F64, F64, I32, I32] -> [F64]
    "math.Log10": [F64, I32, I32] -> [F64]
    "math.Log2": [F64, I32, I32] -> [F64]
    "math.Remainder": [F64, F64, I32, I32] -> [F64]
    "math.Sinh": [F64, I32, I32] -> [F64]
    "math.Cosh": [F64, I32, I32] -> [F64]
    "math.Tan": [F64, I32, I32] -> [F64]
    "math.Tanh": [F64, I32, I32] -> [F64]
    "crypto/sha1.block": [I32, I32, I32, I32, I32, I32] -> []
  Memories:
    "memory": not shared (2 pages..)
  Tables:
  Globals:
```
Targeting wasi, and there's still the imports for `time.startTimer` and `time.stopTimer`. A little lost where to go from here. Looked through the IR before the error, and unable to extract anything meaningful. Without giving access to the code itself is there anything you can point me to to start digging?

cuongvo, 2021-11-07:
Ah, it's not implemented per https://github.com/tinygo-org/tinygo/issues/1037 There's a PR that was never merged linked https://github.com/tinygo-org/tinygo/pull/1402

dkegel-fastly, 2021-11-14:
Suggest closing this as a dup of #1037....

closed 2021-11-14 by cuongvo