tinygo-org/tinygo #2240: Arduino Uno: time.Now() not increasing

jakub-m, 2021-11-06:
The following code on Arduino Uno does not work as expected. I expected that the led will be HI because `time.Now` advances. Instead, it stays LO, because, apparently, time.Now does not advance.

Any clue?

On a side note, `time.Sleep` seems working fine.

```
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	led.Low()
	start := time.Now()
	for {
		if time.Now().After(start) {
			led.High()
		}
	}
}
```

soypat, 2021-11-08:
I had a similar problem a while back, apparently `time.Now()` is very limited on AVR boards (Arduino UNO, Mega...). It only records the amount of time elapsed during `time.Sleep`. So if your program has a `time.Sleep` it should hopefully work better.

jakub-m, 2021-11-09:
Do you know where I can find implementation details? I grep-ped tinygo source but I couldn't find how `time.Now` is implemented for avr controllers.

soypat, 2021-11-10:
`src/runtime/runtime_avr.go` contains `func ticks() timeUnit`  which returns the package global `currentTime`. This constant is only modified by `sleepTicks(d timeUnit)`, which is called by `time.Sleep()`.  Ideally these should be implemented with hardware like arduino's `millis()` function

aykevl, 2021-11-10:
This is a duplicate of https://github.com/tinygo-org/tinygo/issues/952 and is related to https://github.com/tinygo-org/tinygo/issues/1792.

We could start to use the built-in timers, but that will interfere with their usage for PWM (servos, audio output, etc). The Arduino environment decided to use one of the timers for timekeeping, making it difficult to use together with PWM.
It's a tradeoff.

jakub-m, 2021-11-10:
All clear, awesome. Thanks!

closed 2021-11-10 by jakub-m